Ultimaker/Cura #12223: (Cura 5.0.0 Beta) Slicing Failed

rwm002s, 2022-05-17:
### Application Version

5.0.0 Beta+1

### Platform

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

Imported a personal design.
Clicked "Slice".

### Actual results

"Slicing failed with an unexpected error"

### Expected results

Expected the file to be sliced
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8704495/cura.log)
[dpool card 1.zip](https://github.com/Ultimaker/Cura/files/8704500/dpool.card.1.zip)
.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No further information.

fvrmr, 2022-05-18:
Hi @rwm002s thank you for your report.
I can also reproduce this in 5.0 stable which we released yesterday. I will bring this up with the team. Keep you posted!

Ampule2, 2022-05-18:
Full release 5.0.0 also producing Slice Errors in Windows 10

[neck.zip](https://github.com/Ultimaker/Cura/files/8716159/neck.zip)
[OOIpvHIyfG.zip](https://github.com/Ultimaker/Cura/files/8716167/OOIpvHIyfG.zip)

fvrmr, 2022-05-18:
Turns out this is related to https://github.com/Ultimaker/Cura/issues/12019 so I will close this issue.
I added your file as a test case!

@Ampule2 please attach a project file (file - save project) in a new issue, to check if it is related!