Ultimaker/Cura #12226: ultimatker-cura.exe flagged with virus

mzac, 2022-05-17:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

n/a

### Reproduction steps

Trying to install Cura 5.0 and my anti-virus software is flagging it as having a virus with the detection of: Gen:Variant.Tedy.68373

My system is using Cisco Secure Endpoint as a file scanner as my computer is corporately managed.

### Actual results

Cura 5.0 cannot be installed

### Expected results

It should not contain a virus.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

mak0t0san, 2022-05-17:
Multiple vendors have flagged this as malicious.  Here's the results from VirusTotal: https://www.virustotal.com/gui/file/2cba56016f1ea0a285a9b3b8bd7abf9323feaeb0948b11d5094d72115da21271

[image]

nallath, 2022-05-17:
Thanks for reporting this! We did upload it to virus total, but we didn't get any results from it. We're looking into this right now. 

It's probably a false alarm, but it's always good to err on the side of caution with these kind of things.

mzac, 2022-05-17:
@nallath  You don't get any report on the main download package but on the embedded exe of Ultimaker-Cura.exe.

To test, I just downloaded it again inside a windows Sandbox environment, extracted and re-uploaded to virus-total and still get the same warnings.

[image]

KodeMunkie, 2022-05-17:
Since this could actually be a virus (e.g. hack of Github of exploiting the fact it's a big release) I attempted to rollback. 
I uninstalled 5.0.0 and tried installing 5.0.0-beta+1 again but now nothing works. 
This is looking more like a virus if I can't downgrade the version.

mak0t0san, 2022-05-17:
> Since this could actually be a virus (e.g. hack of Github of exploiting the fact it's a big release) I attempted to rollback. I uninstalled 5.0.0 and tried installing 5.0.0-beta+1 again but now nothing works. This is looking more like a virus if I can't downgrade the version.

Most likely just a false positive.  Looks like Cura uses PyInstaller in the executable since Cura runs Python, and this triggered a false positive.

nallath, 2022-05-18:
>You don't get any report on the main download package but on the embedded exe of Ultimaker-Cura.exe.

I know. We also tested the embedded exe. 

> This is looking more like a virus if I can't downgrade the version. 

I think it's likely something else. We've been having false positives for virus scanners for the entire lifetime of Cura. We've had a few releases that didn't flag as a virus, but those are rare. Unfortunately, there is little we can do but wait now that we've uploaded this as a potential false positive. We have checked in so far that we could and have not identified any malicious infection (but that obviously doesn't mean that we're right about that)

ArnoldHenri, 2022-05-18:
Same issue for me since yesterday. Restaured exe file thinking it wasa false positive but now receiving fake update messages from my ASUS computer which seems to be related to an old hack on ASUS servers...coincidence? I will uninstall it till you can proove it is not a virus... Hope it was not breaking something on my machine

nallath, 2022-05-18:
Our investigation now points to us changing the icon of the application as being the culprit. Yeah, you read that right. The icon...

KodeMunkie, 2022-05-25:
Any update? This is potentially affecting many of your customers with a major publicised release so I'm surprised that you've not verified and confirmed here the released checksums against your build server with a fresh build and/or notified Bitdefender et al (who from experience usually sort it in 48 hours).

HermanGaviriaMarin, 2022-05-26:
McAfee is also treating Cura 5.0 as a threat, though turning off real time scanning temporarily works it is something that should be fixed or addressed. Updates on the situation are much appreciated.

Ghostkeeper, 2022-05-27:
To be clear, Cura is open source and so are all of its dependencies. We know that our source code (as you see it here on Github) is under our control, since the commit IDs didn't change and are regularly signed by us. We haven't had any known breaches of our build servers and we also re-create our build servers quite regularly as of late from Terraform due to workflow improvements we were making. So we're confident that there is no actual virus in there.
The build can be verified by externals. Although it's unlikely that you get the same hash due to timestamps if you also build cura-binary-data (not to mention differences in compilers, Glibc and other runtime libraries on different systems), they should give the same results in a virus scanner.

Before the beta release we ran this through VirusTotal and found that three of them failed. Apparently there is [an issue](https://stackoverflow.com/questions/43777106/program-made-with-pyinstaller-now-seen-as-a-trojan-horse-by-avg) with PyInstaller (which is used since 5.0 to create a binary out of our Python code). We made [a partial fix](https://github.com/Ultimaker/cura-build-environment/pull/133), but still one vendor had a false positive.
After the beta release we fixed an icon being corrupt on MacOS. Apparently the pixel data in the new icon looked a lot like this "Tedy" variant 68373, causing AI-based detectors to trip. We reported the binary as a false positive to several vendors. That same link posted above (same hash, as you can see) now only shows one hold-out.
[image]
We just uploaded it as a false positive with SecureAge too. They are reviewing it now. They are saying that it needs at least 48 hours.

Ghostkeeper, 2022-06-01:
Also SecureAge now finds this binary to be safe now. Same link as what Mak0t0san posted above: https://www.virustotal.com/gui/file/2cba56016f1ea0a285a9b3b8bd7abf9323feaeb0948b11d5094d72115da21271

So we'll consider this issue fixed. If an anti-virus software listed here still reports it as dangerous, please update your virus definitions!

closed 2022-06-01 by Ghostkeeper