Ultimaker/Cura #12227: [5.0 final] Inconsistent print settings selection behaviour

ungutknut, 2022-05-17:
### Application Version

5.0

### Platform

Windows 10, Xubuntu 21.04

### Printer

doesn't matter

### Reproduction steps

1. Click on the textbox at "infill density" (just one random example; it happens on all of the print settings textboxes) to change the value. The cursor will appear after the caption in the textbox. This is the default behaviour.
2. Now use the vertical scroll bar - not the scroll wheel! Really click and drag the scroll bar.
3. Click on a textbox again (e.g. "infill density"). This time the caption gets selected - which IMHO should be default behaviour anyway.

### Actual results

Inconsistent selection behaviour.

### Expected results

Consistent selection behaviour.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I also described the issue at the 5.0 beta feedback [thread](https://github.com/Ultimaker/Cura/discussions/11842) but somehow most of my posts there keep vanishing for unknown reasons.

fvrmr, 2022-05-18:
Hi @ungutknut thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

Auntystu, 2022-05-19:
I have found this BUG. selecting 'any' settings box results in selection of setting, however, with the first key press the number selected is then taken as 'gospel' and no longer selected. eg. Select Wall Thickness, delete current setting of 1.2mm, press 1 and it selects 1.0mm automatically and no longer is selected, to continue, you have to select Wall Thickness again, delete the 0 and continue with another number, this also happens with infill % i have to select infill 3 times to enter 100%,making the a very labourious ant irritating task

fvrmr, 2022-05-19:
I created a ticket for this, for internal reference see CURA-9320

MariMakes, 2023-07-31:
Hey @ungutknut,

Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue with reducing the infill to zero resulting in unexpected behavior and it seems to me that it's a duplicate of https://github.com/Ultimaker/Cura/issues/12227

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12227

closed 2023-07-31 by MariMakes