Ultimaker/Cura #12228: Slicing failed after changing Layer Height from 0.12 to 0.15mm

Nikolay-Po, 2022-05-17:
### Application Version

5.0.0

### Platform

Debian Linux 11

### Printer

Anycubic Mega X

### Reproduction steps

Open project, import as project. Try to slice. Slicing is OK. Then try to change layer height to 0.15mm (0.13mm and other heights falling too).
[Silica Gel Box.zip](https://github.com/Ultimaker/Cura/files/8709410/Silica.Gel.Box.zip)

### Actual results

With 0.15mm layer heigth, an unexpected error is occurring. Can't slice any height except 0.12mm.

### Expected results

Expected to be sliced with any reasonable layer height, say between 0.05 to 0.15mm.
[cura.log.zip](https://github.com/Ultimaker/Cura/files/8709400/cura.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Nothing unusual. In any other aspectrs Cura 5.0.0 works for me.

GregValiant, 2022-05-17:
It looks to be something in the settings/overrides within the "ABS N0.3 L0.12 LA0.07" profile.  When I re-select that profile and discard all the changes then the model slices regardless of the layer height.

fvrmr, 2022-05-18:
Hi @Nikolay-Po thank you for your report.
This is a duplicate of an issue we have on our backlog. That is linked to https://github.com/Ultimaker/Cura/issues/12019, I will add your project file as a test case to our ticket, but close this issue!

closed 2022-05-18 by fvrmr