Ultimaker/Cura #12229: Can't open or save files in Linux AppImage version

sputjds, 2022-05-17:
### Application Version

5.0.0

### Platform

Ubuntu Mate

### Printer

Anycubic Mega S

### Reproduction steps

After launch the appimage (linux) version of Cura 5.0 I click of File, Open and nothing happens. I click on the folder icon and nothing happens. I can drag an stl file and drop it and it will open, but when I click on Save to File, nothing happens. 
I can click on File, Open Recent, and select a recent file and it will open fine.

### Actual results

Nothing happens when I click of File, Open, or File, Export, or File, Export Selection, or File Save Project, or if I click on the Save to File button. 

### Expected results

These actions should open a file save or file open dialog, but they don't.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8709706/cura.log)

ungutknut, 2022-05-17:
I can  confirm this issue for Xubuntu 20.04. Workaround is to drag & drop files onto the build plate. But saving a project isn't possible unfortunately.

I've already described the bug in the 5.0 beta [feedback thread](https://github.com/Ultimaker/Cura/discussions/11842#discussioncomment-2620841) but obviously there was no one else with this issue. My random guess is that our installed QT versions might be outdated since it works fine on Xubuntu 21.04 with a more recent QT version. Yet I actually believe that the appimage would ship it's own QT version.

What's your Ubuntu and QT version?

sick4noise, 2022-05-18:
I have this same issue on Gentoo using QT 5.15.3. I would try upgrading Qt but 6.3 is not supported by Gentoo yet
[cura.log](https://github.com/Ultimaker/Cura/files/8712326/cura.log)
.

ungutknut, 2022-05-18:
> I don't know how to tell what version of QT I have

You can try it with `dpkg --list | grep -i libqt`

Probably most of the listed packages will be of version 5.12.8 or something. On Xubuntu 21.04 where it works I have 5.15.2
But I have no idea if this is even linked to the issue in any kind. Just a random guess.

Nevertheless I'm pretty sure a `sudo do-release-upgrade` to upgrade your system to Mate 22.04 LTS would solve the problem... and perhaps raise new issues.

chipmunk1977, 2022-05-18:
This appears to be the same issue I raised:  https://github.com/Ultimaker/Cura/issues/11874
It doesn't seem to be affecting all Linux versions and there is a workaround, in as far as you can drag and drop files, and you can save to a removable drive.

sputjds, 2022-05-18:
Okay, so I have QT version 5.12.8. I will attempt upgrading to Mate 22.04 and see what happens. 
As for the workaround of dragging and dropping files, my post clearly states that I can do that, and if I have a USB stick plugged in, I can save the sliced file to that drive (thanks chipmunk1977 for telling me about that workaround), but it doesn't allow me to name the file - it just saves it directly to the drive. So, yeah, I could use the workaround, but I prefer to save the gcode to my hard dirve and I'd prefer to have the file dialog actually work like it should.

sputjds, 2022-05-18:
THIS IS NOT FIXED. Closing this issue because others have already reported the same issue and I really don't feel like dealing with it any more. I'll stick with Cura 4.13 and Prusa Slicer.

closed 2022-05-18 by sputjds

ungutknut, 2022-05-19:
> So I give up. Maybe I'll try again when v5.1 comes out

You can give that workaround a try though (last paragraph):
https://github.com/Ultimaker/Cura/issues/12241#issuecomment-1130753983

sputjds, 2022-05-19:
Not gonna bother. Ultimaker can do their own beta testing.