Ultimaker/Cura #12232: Sign in button does not work.

erickwill, 2022-05-17:
### Application Version

5.0.0

### Platform

Arch Linux Rolling

### Printer

Tronxy X5SA

### Reproduction steps

Try to Sign in using Arch Linux or other distribution.

### Actual results

It does not open any web browser allowing the log in process.

### Expected results

It should open a web browser and allow the log in process. I tried with multiple web browsers.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8710373/cura.log)

erickwill, 2022-05-18:
> What are you using as a desktop environment (KDE/Gnome etc.)?
> 
> Can you try any other external link, such as the links under the app switcher menu?

I'm using Gnome. I also tried to open external link on Cura and it did also failed. I tried in 3 different laptops running Arch with similar configurations and same results.

fvrmr, 2022-05-18:
Are you using the AppImage?

erickwill, 2022-05-18:
> Are you using the AppImage?

Yes, I'm using the appimage.

jellespijker, 2022-05-19:
@MariskaMaas  FYI

fvrmr, 2022-05-19:
We discussed this with the team and will create a ticket for this.

erickwill, 2022-05-19:
Thanks! Looking forward for a fix.

slabua, 2022-07-09:
Still no go? I'm stuck with v4 because of this issue.

fieldOfView, 2022-07-10:
> I'm stuck with v4 because of this issue.

You don't *have to* sign in to use Cura.

slabua, 2022-07-10:
> > I'm stuck with v4 because of this issue.
> 
> You don't _have to_ sign in to use Cura.

Uh? I don't even have a Cura account.
All links are not working, plugins, update checker, everything that would open an url in the browser just does nothing, making v5 completely unusable for my workflow.

slabua, 2022-07-10:
Also not sure why issue 12527 has been closed in favour of the present one, since 12527 is actually the more general case.

fieldOfView, 2022-07-10:
Sorry, this issue is specifically about the sign in button, and you did not mention (before, in this issue) that you were having issues with other buttons.

I guess the other issue was closed as a duplicate because it shares the same underlying problem that a browser window is not opening.

slabua, 2022-07-10:
I see. I commented here because the other one was closed.
Then I guess it shouldn't have been closed as duplicate of this.
Waiting for a fix.

MariMakes, 2022-08-22:
Hey all.  

This should be fixed in Ultimaker-Cura-5.1.0-linux-modern.AppImage.asc
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.1.0

Thanks for reporting!

PR #12736 (2022-07-12): Cura 9205 appimage not launching on ubuntu 2204
Building two AppImages, one based on Ubuntu 20.04 and one with on Ubuntu 22.04. This also solves the sign in button not working on some Linux systems. Keep in mind that this is more of an intermittent solution and that we should still try to solves this in a more generic/future proof-way

Fixes both CURA-9342 and CURA-9205

The results of this run can be found here: https://github.com/Ultimaker/Cura/actions/runs/2648505664

- Fixes #11856
- Fixes #12453
- Fixes #11900
- Fixes #12232
- Fixes #12291
closed 2022-08-22 by MariMakes