Ultimaker/Cura #12234: Could not probe OpenGL.

erickwill, 2022-05-17:
### Application Version

5.0.0

### Platform

Arch Linux Rolling

### Printer

Tronxy X5SA

### Reproduction steps

Try to run cura.appimage with intel video card.

### Actual results

Try to run cura.appimage with intel video card and a pop up comes out:
This program requires OpenGL 2.0 or higher. Please check your video card drivers.
I tried in 3 different laptops, both with modern intel video hardware.

### Expected results

To run smoothly as the version 4.13 and early.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[erick@archlinux Apps]$ ./Ultimaker-Cura-5.0.0-linux.AppImage 
`
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_Ultimari5YfP/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_Ultimari5YfP/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
`

Pop up message:
`Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.`

fieldOfView, 2022-05-17:
Could be a duplicate of https://github.com/Ultimaker/Cura/issues/11856

Dorro101, 2022-05-18:
Confirm same error on Lenovo ThinkPad X1 Tablet Gen 3, Intel® Core™ i5-8250U, Mesa Intel® UHD Graphics 620 (KBL GT2), Fedora 36, Gnome 42, Wayland. It runs from terminal using `LD_PRELOAD=/usr/lib64/libstdc++.so.6 /home/dorian/.local/share/appimages/Ultimaker-Cura-5.0.0-linux.AppImage`. I changed the path from `/usr/lib/x86_64-linux-gnu` to `/usr/lib64`. I created an executable shell script and linked this to a .desktop file so I could execute from the Gnome application try and Dock to Dash.

closed 2022-05-18 by nallath

changedsoul, 2022-07-19:
Same message on a Dell XPS 17" 9720 with Intel Corporation Alder Lake-P Integrated Graphics Controller