Ultimaker/Cura #12237: Changing Infill % curser error

DaveMeade5, 2022-05-18:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender S1

### Reproduction steps

When trying to change to infill %, as soon as I type one number, the curser jumps outside the box, as if i have pressed the enter key.
EG, I try to change 100 to 30
I put the curser in the box, delete 100 and then as soon as I press 3, the curser jumps outside the box and leaves 3%
I then have to go back in to make it 30%
I does this every time
I have only really noticed it on the infill parameter. Others may do it though.
It has never done this on 4.13

### Actual results

See above

### Expected results

I expect the curser to stay in the back until I have made all the alterations I need to.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

ungutknut, 2022-05-18:
I can confirm this issue. Another observation: when typing in a new number, cura scrolls the settings in a random direction. Current workaround for me: type the number very fast.

fvrmr, 2022-05-18:
Hi @DaveMeade5 thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

DaveMeade5, 2022-05-18:
Thanks. 
Glad you worked out what I meant with my atrocious grammar.... I was typing on my phone without my glasses on.

fvrmr, 2022-05-19:
I linked this is an ticket we already had on our backlog. Devs see CURA-9314