Ultimaker/Cura #12241: OpenGL 2.0 error at startup

guyvia, 2022-05-18:
### Application Version

5.0

### Platform

Ubuntu 22.04

### Printer

Wanhao Duplicator i3+

### Reproduction steps

Run the Appimage file

### Actual results

Error stating that OpenGL 2.0 is not present, even though I am running a modern(ish) graphics card is capable of 4.6.

[cura.log](https://github.com/Ultimaker/Cura/files/8717219/cura.log)

### Expected results

It should run.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

.

erickwill, 2022-05-18:
I have the same issue in one of my laptops, also, with a modern'ish' intel graphic card.

PowerWiesel, 2022-05-18:
I got a similar issue

```
sudo apt list --installed | grep amdgpu

libdrm-amdgpu1/testing,unstable,now 2.4.110-1 amd64  [Installiert,automatisch]
libdrm-amdgpu1/testing,unstable,now 2.4.110-1 i386  [Installiert,automatisch]
xserver-xorg-video-amdgpu/testing,unstable,now 22.0.0-1 amd64  [installiert]
```

```
    OpenGL version string: 4.5 (Compatibility Profile) Mesa 21.3.8
    Vendor: AMD (0x1002)
    Device: AMD STONEY (DRM 3.44.0, 5.17.0-1-amd64, LLVM 13.0.1) (0x98e4)
    Version: 21.3.8
    Accelerated: yes
    Video memory: 512MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
```
```
5.17.0-1-amd64 #1 SMP PREEMPT Debian 5.17.3-1 (2022-04-18) x86_64 GNU/Linux
```
Output while starting
```
./Ultimaker-Cura-5.0.0-linux.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```
No log was generated.

Starting Cura with `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage` seems to work as a workaround.

guyvia, 2022-05-19:
Running it from command line that with the LD_PRELOAD gives its own issues. When i try to sign in, it will not open a browser and the terminal running it shows:

/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_tls_channel_binding_error_quark
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
mkdir: /tmp/.mount_UltimaiqptfP/libselinux.so.1: no version information available (required by mkdir)
/usr/bin/google-chrome-stable: symbol lookup error: /lib/x86_64-linux-gnu/libpango-1.0.so.0: undefined symbol: g_memdup2

PowerWiesel, 2022-05-19:
> I also am unable to open an STL when running it that way, so... I don't know what gives here.

opening and slicing STL and 3MF works for me

rjsc, 2022-05-20:
Same problem here, on kubuntu 22.04.
In fact it's happening since the 5 beta version and has not been fixed.

nallath, 2022-05-20:
Duplicate of #11856

closed 2022-05-20 by nallath