Ultimaker/Cura #12244: Cura 5.0 interface is a total regression on Mac. No native menus anymore!

tokamac, 2022-05-18:
### Application Version

5.0

### Platform

Mac

### Printer

Prusa i3 MK3S

### Reproduction steps

Cura menus are not located inside the Mac unified menu bar anymore (top of the screen). Instead they appear as a Windows app on a PC (or a WINE emulated port, or a crude X11 interface on a Mac) i.e. the menus being located inside the window itself. Even the fonts there are off and not Mac-friendly anymore. They look non-native and terrible as if I was running Windows instead of macOS. Look at this:

All Cura versions before 5.0 on Mac:

[image]

### Actual results

Now, here is Cura 5.0 on Mac now:

[image]

### Expected results

Please fix this. If you work on a Mac, it's not to have applications that seem half-baked, especially as the prior Cura interfaces were fine for years.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Terrible UI experience overall. Even Cura Arachne Public Betas didn't have such user interface issue, introduced in the 5.0b branch.

Guilouz, 2022-05-18:
Confirmed, same here.

Vandresc, 2022-05-27:
Hi @tokamac, the issue appeared after the QT upgrade we did for 5.0. We tried for a fix before and during beta but failed to find an easy elegant one. Also due to all other priorities we have, we have decided to defer this issue for now. Thank you for your feedback!

tokamac, 2022-05-27:
You're welcome @Vandresc so I'll throw Cura in the trash and go back to PrusaSlicer which is blazingly fast, 100% compatible with Apple Silicon now, and above all designed respectfully for everyone including Mac Users. Ciao!

YassineGherbi, 2022-08-10:
@Vandresc is there any estimation on when we can expect this? I hope this will be a priority. It really makes the app so much worse, despite all the new improvements. It's just not worth it using Cura 5 at this point on a Mac

derekcat, 2022-10-11:
+1 I'd really love to see this resolved.
It's a huge put off when it's acted as a proper Mac application up until now :( 

Also thanks for all the work on Cura in general!

sammcj, 2023-01-24:
Happy to test beta builds that aim to improve this if that helps.

lucianostegun, 2024-09-14:
It's terrible when apps in macos don't have native menus.
It's even worse when an app has a native menu and it gets removed in newer versions.
That actually made me downgrade to 4.x

nanoant, 2025-03-15:
It is ridiculous that Cura website shows macOS with proper menu, while in reality the same app does not work as shown on the website.

## Website

[image]

## Reality

[image]

inventor7777, 2025-04-23:
I really hope this gets improved soon. I tried updating from 4.8 and when I saw the UI of 5.9 I instantly went back to 4.8. Really dumb that it still looks this bad after almost 3 years.