Ultimaker/Cura #12247: Mr Ian Davys

iandavys, 2022-05-18:
### Application Version

5.0.0

### Platform

windows 11

### Printer

Flsun Q5

### Reproduction steps

I
[image]
[Cura failed project.zip](https://github.com/Ultimaker/Cura/files/8720469/Cura.failed.project.zip)

### Actual results

I gave up and i loaded my Cura from flsun 1.4.15 (that came with the Q5) yep it worked and now all is good.
it would be very intresting to find out why the new Cura never worked.
Hope you can help.

thanks from Ian

iandavys@vivaldi.net

### Expected results

It working and slicing my small project for someone.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Sorry think the photo should of gone in this space!

Thanks.

from Ian
[image]

GregValiant, 2022-05-18:
I'm not sure what happened.  I changed to the "Draft" profile, discarded all the changes, and then switched back to the "Extra Fast" profile and it sliced.
This problem with 5.0 has been noted before.

Vandresc, 2022-05-30:
Thank you, @iandavys. I will bring this up with the team. We indeed have some cases where the slicing still fails.

Vandresc, 2022-05-31:
Devs see CURA-9345 for our internal reference.

closed 2024-04-05 by nallath