Ultimaker/Cura #12248: Cura 5.0.0 does not retain screen size/position when  starting

MarkALynch, 2022-05-18:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 5 Plus, custom

### Reproduction steps

When I started Cura prior to version 5.0 release, it would open at the previous size and location I had close it from. I have verified that the Cura.cfg file contains the right Height and Width settings, and they change if I resize the application, but it always starts in a fixed size that is not near what I was running it at. 

I have three screens, each screen is set for 1920x1080. I run with the main display as the center screen.

It does not seem that the cura.cfg file has a display or valid window_left setting as it's not accounting for the screen or the absolute left position.

Also, the starting Windows height and width seem to be a default, instead of the values that are saved when I close cura after resizing.

The Cura splash screen also starts on the far left monitor and I can't move it.

Cura 4.13.1 still retains the screen position and size settings when changed. 

### Actual results

last_run_version = 5.0.0
accepted_user_agreement = True
window_height = 985
window_width = 1845
theme = cura-dark
version = 7
window_left = 67
window_top = 37

### Expected results

Cura to start on the same screen and at the same size as it was closed.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8721008/cura.log)

2022-05-18 17:31:49,078 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [87]: Adding QT6 plugin path: C:\Program Files\Ultimaker Cura 5.0.0\PyQt6\plugins
2022-05-18 17:31:49,080 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [97]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 5.0.0\PyQt6\plugins
2022-05-18 17:31:49,082 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-05-18 17:31:49,084 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 5.0.0
2022-05-18 17:31:49,085 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 8.0.0
2022-05-18 17:31:49,087 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-05-18 17:31:49,089 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib\uranium must be a folder.
2022-05-18 17:31:49,091 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib64\uranium must be a folder.
2022-05-18 17:31:49,092 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\lib32\uranium must be a folder.
2022-05-18 17:31:49,094 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Ultimaker Cura 5.0.0\plugins must be a folder.
2022-05-18 17:31:49,095 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.
2022-05-18 17:31:49,097 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.
2022-05-18 17:31:49,099 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-05-18 17:31:49,100 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is \\VS2019STD-CITS\RedirectedFolders\mark.lynch\AppData\Roaming\cura\5.0
2022-05-18 17:31:49,102 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is \\VS2019STD-CITS\RedirectedFolders\mark.lynch\AppData\Roaming\cura\5.0
2022-05-18 17:31:49,104 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is C:\Users\mark.lynch\AppData\Local\cura\5.0\cache
2022-05-18 17:31:49,105 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\bundled_packages\cura.json
2022-05-18 17:31:49,107 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\bundled_packages\uranium.json
2022-05-18 17:31:49,108 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2022-05-18 17:31:49,109 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[... 183 lines cut ...]
2022-05-18 17:31:54,861 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\overhang.shader]...
2022-05-18 17:31:54,865 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\striped.shader]...
2022-05-18 17:31:54,868 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\transparent_object.shader]...
2022-05-18 17:31:54,871 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\striped.shader]...
2022-05-18 17:31:54,874 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\xray.shader]...
2022-05-18 17:31:54,878 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\xray_composite.shader]...
2022-05-18 17:31:54,881 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\shaders\platform.shader]...
2022-05-18 17:31:54,888 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\xray.shader]...
2022-05-18 17:31:54,900 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [58]: Updating QualityProfilesDropDownMenuModel.
2022-05-18 17:31:54,905 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2022-05-18 17:31:55,053 - DEBUG - [JobQueueWorker [0]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2022-05-18 17:31:55,450 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\uranium\resources\shaders\default.shader]...
2022-05-18 17:31:55,454 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\shaders\grid.shader]...
2022-05-18 17:32:04,328 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2022-05-18 17:32:04,332 - INFO - [MainThread] cura.CuraApplication.closeApplication [614]: Close application
2022-05-18 17:32:04,335 - INFO - [MainThread] cura.CuraApplication._setLoadingHint [807]: Initializing Active Machine...
2022-05-18 17:32:04,382 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2022-05-18 17:32:04,385 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2022-05-18 17:32:04,390 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2022-05-18 17:32:04,392 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2022-05-18 17:32:04,436 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:237:25: Unable to assign QString to QRegularExpression
2022-05-18 17:32:04,446 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [495]: Shutting down cura
2022-05-18 17:32:04,480 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [354]: Attempting to kill the engine process
2022-05-18 17:32:04,482 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Killing engine process
2022-05-18 17:32:04,499 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [363]: Engine process is killed. Received return code 1

Vandresc, 2022-05-30:
Also reported here https://github.com/Ultimaker/Cura/issues/12102. Issue will be re-discussed with the team.

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

- A surprising connection was found to some particular audio drivers: https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232
- Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

MarkALynch, 2023-03-31:
> Quick update on our side,
>
> We might have a fix for some of the issues reported here. 🎉
> It's a bit hard for us to test, but we have a nightly build available.
> Can you please try it out and let us know if it resolves your issue?
>
> You can find the download link and discussion here. <https://github.com/Ultimaker/Cura/discussions/15098>

Installed stand alone, Win 10 x64, latest updates

3 x Screens, Screen layout 2 1 3, all screens 1920 x 1080 resolution.

Cura splash starts on screen 1 – works fine, as it should. Previously it was starting on Screen 3 instead of Screen 1. Splash should start on Screen 1 – good
GUI size, also started on Screen 1, size was the same as I had last stretched (5.3.x version).
Closed GUI, opened, screen size retained. Changed GUI size and moved to screen 2, closed gui. Opened Cura again, opened properly on Screen 2 and remembered the size – good.
This version looks like it’s seeing the proper screen number and size/top/left position – thanks bunches.
Great work, you and the team deserve a big Bravo!

One thing, and I know you’re not working on this. I just upgraded to a Sonic PAD running Klipper and learned that my hot-end can’t keep up with the needed volumetric flow when running larger nozzles (.6mm or .8mm). I had been using Cura for my 5 Plus exclusively, using Prusa for my 3 Pro exclusively. Cura doesn’t included a Max Volumetric Flow setting, so I had to switch to Prusa to get the overall MVF limit. I did purchase a new hot end, a Spider V3 and it can handle just over 24mm3/sec flow, but even with that, running a .6mm nozzle, .63mm line width, and .36mm layer height, it can outrun the hot end melt rate without a Max Volumetric Flow limit in the slicer. Do you know if anyone is planning on adding MVF to Cura now that it’s common to have high-flow hot ends and larger nozzles?

Any time you need a version/feature tested, please feel free to reach out to me and I’ll be more than happy to help.

Sincerely,
Mark Lynch

rburema, 2023-04-04:
@MarkALynch Thanks for testing! It's good to hear it works for at least one person.

With regards to the Max Volumetric Flow setting, there is an internal ticket for it (though that doesn't always say much), and a community member made a PR for it: https://github.com/Ultimaker/Cura/pull/14452 Additionally, a rather verbose discussion going back to the beginning of 2019 can be found here: https://github.com/Ultimaker/Cura/issues/5248 All in all, it looks like it will be coming _eventually_ one way or the other. Wheter that is in time for 5.4 (or even later) remains to be seen.

MarkALynch, 2023-04-04:
Thanks for the feedback.

With the game changer of the Sonic Pad, at least for Creality, while I started programming in the late 70’s and have spent decades designing electronic hardware, I really didn’t want to setup a pi and klipper and get into the changes to my printer. The Sonic Pad made it simple, and to be honest, while I’m OCD about creating filament profiles, the Sonic Pad improved my print quality over what I already thought was the best I could get.

The problem I ran into was that with the SP, Klipper, tuned, moving the ACCEL to 5000-7000 and Jerk to 15, and then setting the speed to 120mm/sec, I found that I was outrunning the ability to melt filament fast enough with a .6mm or .8mm nozzle – and I had switched to a Gulf Coast Robotics all metal hot end about 2 years ago. What I found is that the Factory and GCR hot ends can’t really do more than 10mm3/sec flow rate, but the Spider V3 can easily maintain 24mm3/sec flow rate. I tried to adjust Cura to get optimal speeds, but without MVF it was impossible to not get extruder clicking as the hot end was unable to melt filament that fast. In Prusa, they have had a MVF for a long time, even before getting the Spider V3 hot end, still just running the GCR hot end, I set the MVF to 10mm3/sec and was able to print faster, even setting speeds to 180 worked as everything was limited to 10mm3/sec regardless of the line type, line width, etc… With my 3 Pro, not using the Sonic Pad, I’m still using Cura, but with the 5 Plus and Sonic Pad, I have to use Prusa to get the speeds it’s capable of – and I also set Prusa to use ACCEL/JERK for time estimates only, to NOT output ACCEL/JERK in g-code, using the values in the Sonic Pad instead.

Thanks for fixing the position and scale issue with Cura.

Mark Lynch