Ultimaker/Cura #12249: skirt is not copy model edges

vitasrutek, 2022-05-19:
### Application Version

5

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. New project
2. Add new STL file (caliper) - [](url)https://www.thingiverse.com/thing:1661962
3. Set all settings default (instead quality set to 0,12 mm)
4. Sliced it
5. Skirt in one place not copy edges of imported model
6. 
[image]

### Actual results

Tried to print it and bug is printed too :)

### Expected results

Should be correct edge copy of skirt.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_simple_calipers_with_depth_gauge.zip](https://github.com/Ultimaker/Cura/files/8723314/CE3_simple_calipers_with_depth_gauge.zip)

Asterchades, 2022-05-19:
The skirt only goes around what's on the first layer. The part that isn't being covered by the skirt has a very slight (0.1mm) shelf under it, which is being maintained by the slicer due to the Slicing Tolerance set. As a result that piece is not present on the first layer and the skirt is behaving properly. You can see this more clearly by dragging the slider down to only show the first layer.

You can either change the Slicing Tolerance to "Inclusive" to force the layer to be printed based on the 0.02mm portion that exists within your first layer, or switch to a 0.2mm or higher Initial Layer Height such that the gap is outside of the default "Middle" tolerance.

Vandresc, 2022-05-31:
Hi @vitasrutek , thank you for your feedback. @Asterchades is correct, in this case, this is expected behavior, and the workaorund should do the trick. Thank you both!

closed 2022-05-31 by Vandresc