Ultimaker/Cura #12253: opening props

AceTanner, 2022-05-19:
### Application Version

5.0

### Platform

windows 11

### Printer

ender 3 pro

### Reproduction steps

deleted and re down loaded cura 5.0

### Actual results

failed to install right

### Expected results

faster printing
[image]
[image]
[image]
 time would be very nice but that is not the case for me 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i would like to have the new cura woring but for some reason it is not

AceTanner, 2022-05-19:
I have just uninstalled Cura 5.0 and deleted it from the computer and the last few months of old files then re-download and installed the Cura 5.0 and still getting the add pop up that will not restart as it should and the printer time I have looked at all setting in both Cura's and the setting are the same so why is Cura adding 4 hours to a 3 hour one wall and 10 % infill with 4 steps before the roof in 1.5mm 2 bottom layers and 4 top layers

nallath, 2022-05-20:
Without log files, there isn't anything we can do to fix the first issue that you reported.

As for the time difference; the slicing algorithm is fundamentally different from what Cura 4.13 was doing. We don't have Creality or ender printers, so we haven't been able to test these printers (or provide for profiles that work well). We have tried to provide some best effort profiles, but I assume that a lot can still be improved on those. Since arachne is so different, it's entirely possible that certain models will take longer to print.