Ultimaker/Cura #12255: Cura 5, No Beep and End of Printing

Toddi3D, 2022-05-19:
### Application Version

5.0

### Platform

Windows 10 64

### Printer

Anycubic Vyper

### Reproduction steps

Sliced the 20mm Cube and print
before i had nothing in the End Code for the Beep and the end of printing and the Beep comes out

With Cura 5 no Beeps

M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M300 S1318 P266
M84 ;steppers off
G90

### Actual results

nothing

### Expected results

nothing

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8727817/cura.log)

GregValiant, 2022-05-20:
The Beep is M300.  It's in your End Gcode in the Vyper definition file so it isn't something that Cura adds to any file.
If it's missing from your End Gcode you can add it back in.  It's the last line.
**M300 S1318 P266**

Your ending gcode also has this line:
"G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more"
The semi-colon in front of the X-20 should not be there.  The line should be:
G1 X-20 Y-20 Z0.5 E-5 F6000 ;
I also suggest removing "speed_travel" and hard code 6000 instead.  The keyword "speed_travel" is in mm/sec and the printer understands mm/minute.  Instead of F6000 (100mm/sec travel speed) it's getting F100 in mm/minute.  Big difference there.

Toddi3D, 2022-05-20:
It is the Endcode that give cura for the Anycubic Vyper. I changed nothing.

Toddi3D, 2022-05-20:
> The Beep is M300. It's in your End Gcode in the Vyper definition file so it isn't something that Cura adds to any file. If it's missing from your End Gcode you can add it back in. It's the last [...]

Its the Endcode from Cura for the Anycubic Vyper. I changed nothing.

GregValiant, 2022-05-20:
Something or somebody changed it.  This is from the End Gcode you posted.  You can see that the M300 line is third from the bottom...
G1 Y180 F2000
M300 S1318 P266
M84 ;steppers off
G90
This is from the End Gcode in the Cura definition file for the Vyper.  Here the M300 line is the very last line.
G1 Y180 F2000
M84 ;steppers off
G90
M300 S1318 P266
So your Ending Gcode had already been altered from what it should have been and did not match the definition file.  That can happen if you open someone else's 3mf project file and choose "Update Printer" instead of "Create New". when prompted.

Toddi3D, 2022-05-20:
> Something or somebody changed it. This is from the End Gcode you posted. You can see that the M300 line is third from the bottom... G1 Y180 F2000 M300 S1318 P266 M84 ;steppers off G90 This is from [...]

I installed cura 5 new and when the M3ü0 line is on the last, the beep not working, so i put it higher and it works. But i changed nothing more, its the standard Endcode from Cura 5.

Vandresc, 2022-06-03:
Hi @Toddi3D ,  We checked the git history and we didn't change anything in the start/end g-code of the anycubic vyper from the moment it was added in August. Therefore it was decided to reject this issue, but feel free to re-open if there are any other related issues to M300 at End of Print. We are glad that you did find a workaround, thanks @GregValiant for that.

closed 2022-06-03 by Vandresc