Ultimaker/Cura #12257: Tree Support partially failed/creted in air

cbrsnoopy, 2022-05-19:
### Application Version

5.0

### Platform

MacOS 12.3.1

### Printer

Tronxy X5SA

### Reproduction steps

Load Model,
Activate Tree Support,
Slice

### Actual results

Tree support Partially generates in the air.
[image]
[image]

### Expected results

Supports are generated like in Cure 4.12.1 as seen here in Picture:
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[TXXPRO2ED01P_Hex3D_BOBobafett_Bust_JetpackTop.3mf.zip](https://github.com/Ultimaker/Cura/files/8729229/TXXPRO2ED01P_Hex3D_BOBobafett_Bust_JetpackTop.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8729232/cura.log)

Distilarouk, 2022-05-19:
i have a similar problem on my ender 3, with 4.13.1, where the tree support comes together it just doesn't. but mine doesn't show in render. only other difference is i'm on a ender 3

[image]

64bittuning, 2022-05-24:
Choose one:
1. Stop using tree supports.  (The support settings (normal) have been greatly improved in the Tronxy profiles. Use of Tree Supports has NEVER been necessary on any model I have ever printed.)

2. Adjust the settings to get your tree supports to work, and then send me a message here on GitHub. I'll work with you to add your changes to the profiles after several other members of the Tronxy community try them as well.

Vandresc, 2022-06-07:
Could be indeed. Will have a look, thank you!
In the meanwhile we made another ticket, devs see CURA-9346 for our internal reference.