Ultimaker/Cura #12260: Search returns too many irrelevant results

Triffen, 2022-05-19:
### Application Version

5.0

### Platform

Linux Manjaro

### Printer

Ender 3

### Reproduction steps

Search for any setting, e.g. example, concentric, fan, brim, skirt. I find that "brim" and "skirt" give different sets of results.
I have Settings guide plugin enabled.

### Actual results

Too many results, with proper results being harder to find
[image]
.

### Expected results

Only settings including the search term are shown.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Settings Guide plugin is enabled, maybe the Search also Searches in the Guide entries. Which is bad idea as there might be all the words there are ;)

nallath, 2022-05-20:
The search field looks through the names of the settings _and_ through the description. The rationale for this is that people will often look for things like "elephant foot", as they want to compensate for that. If it would only search through the setting name, you'd never find that you'd need to change horizontal expansion.

Obviously, the settings guide adds a huge amount of extra text to the descriptions, which also means that the search will suddenly find a lot more settings.

Vandresc, 2022-05-31:
Hi @Triffen this is indeed expected behavior, not a bug but could be a feature request. Thank you for reporting it!

closed 2022-05-31 by Vandresc