Ultimaker/Cura #12262: [5.0] Small external "inside" Brims not correctly generating, interfering with Supports

Asterchades, 2022-05-19:
### Application Version

5.0.0

### Platform

Windows 11 x64

### Printer

Custom FFF

### Reproduction steps

1. Load the model provided.
2. Slice with Supports enabled.
3. Observe

### Actual results

Note the Brim generated around the embossed lettering on the first layer. Instead of generating at the specified size (in this case the default 8mm) it instead has generated only two lines for the most part, with three lines visible under the A. However it is being treated as if the full size has been correctly generated and, again around the letters, no longer generating any Support material on the first layer - be that the brim (**Support Brim** enabled) or just the Support structure.

Toggling **Brim Only on Outside** does not alter this. Curiously it also does not then apply a brim within the completely surrounded hole in the A, either.

Toggling **Brim Replaces Support** does let the Support print in these areas correctly, complete with its own Brim - even within the A - though parts without the Support material still do not generate correctly (only one line present now). However this can cause issues when used in tandem with **Support Horizontal Expansion** where the Brim is forced away from the model in places, reducing (or negating) its adhesion benefit - particularly if the Skirt/Brim and Support Line Widths don't match (eg 0.8mm and 0.4mm respectively).

Reducing the Brim Width such that the full distance will fit within the confines (eg 1.6mm) will cause everything to work as expected - except for within the A.

_NB This behaviour is also present in 4.13.1_

### Expected results

Brims, where present, will be generated correctly. While it's questionable as to whether a Brim should be generated within these letters at all with **Brim Only on Outside** enabled, it should be there without it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[MultiMax_Top.zip](https://github.com/Ultimaker/Cura/files/8732512/MultiMax_Top.zip)

Asterchades, 2022-05-21:
Further information: This is also interfering with Tree Supports. When _Enable Support Brim_ is checked, Trees with too small of a contact area against the bed (which can be a lot of them) simply don't generate an initial layer. The only way to remedy this is to disable the _Enable Support Brim_ option - adjusting the **Support Brim Width**, even down to a single line, will only restore some of the missing first layer elements.

Vandresc, 2022-05-31:
Hi @Asterchades , thank you for your feedback, we have a similar ticket in our backlog.