Ultimaker/Cura #12263: Dark Mode for Mac only. May 19, 2022

mac4kent, 2022-05-19:
### Is your feature request related to a problem?

I am still complain. You did close a ticket too fast without ask me question about Mac. I am lot complain about Too bright light mode. You have to fix everything dark mode through System Mac on cura app.
[image]

Open file cause too bright light mode. You must to add feature through System Preferences > General on Dark real SYSTEM into cura.app
[image]

Please show me your screenshot to proof tell me about (Open FILE)

### Describe the solution you'd like

  See above

### Describe alternatives you've considered

See above

### Affected users and/or printers

See above

### Additional information & file uploads

See above

EtzBetz, 2022-11-06:
Cura already does respect the user's setting for the system controls. Whats missing sadly, is auto dark mode. I created an issue for that.
[image]

mac4kent, 2022-11-06:
Ok can you send me link to get download app only for Mac?

EtzBetz, 2022-11-06:
I've downloaded the latest version from the cura website for macOS

mac4kent, 2022-11-06:
Where?

EtzBetz, 2022-11-06:
https://github.com/Ultimaker/Cura/releases/download/5.2.1/Ultimaker-Cura-5.2.1-mac.dmg

mac4kent, 2022-11-06:
Ok I get from your link and I can see dark mode finally that come support System Setting there is dark mode too.
Let's close a ticket.