Ultimaker/Cura #12264: Crashing when certain popups appear

voxpop9, 2022-05-19:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3V2

### Reproduction steps

Opening Market place or multiply print. 

### Actual results

Market place or multiply print popup doesn't load and cura freezes and stops responding

### Expected results

Market place Market place or multiply print popup should open normally

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8733706/cura.log)

voxpop9, 2022-05-20:
> Does this also happen when opening any of the other popup screens (Preferences, for instance?)

It only happens on certain pop ups. Preferences, for example, opened up fine for me. Seems like Post Processing Plugin screen also crashes. Atm only opening of files and preferences works fine.

pbdroid, 2022-05-26:
This might be the same issue that I am having:
https://github.com/Ultimaker/Cura/issues/12345

These trigger mine crashing:
Attempting to open "Marketplace"
Attempting to open "From Digital Factory"
Attempting to open "Cura Backups\Manage Backups"
When it tries to sync setting with printer

Happened on beta and stable release. I have tried everything. Backing up, complete removal of all versions and config files, running as administrator, getting a working version on another older computer and then moving files over. All to no avail. 

This is a new Alienware laptop running Windows 11.

Do you have the free "Essentials Subscription"?

voxpop9, 2022-05-27:
> This might be the same issue that I am having:
> 
> https://github.com/Ultimaker/Cura/issues/12345
> 
> 
> 
> These trigger mine crashing:
> 
> Attempting to open "Marketplace"
> 
> Attempting to open "From Digital Factory"
> 
> Attempting to open "Cura Backups\Manage Backups"
> 
> Saving a project
> 
> When it tries to sync setting with printer
> 
> 
> 
> Happened on beta and stable release. I have tried everything. Backing up, complete removal of all versions and config files, running as administrator, getting a working version on another older computer and then moving files over. All to no avail. 
> 
> 
> 
> This is a new Alienware laptop running Windows 11.
> 
> 
> 
> Do you have the free "Essentials Subscription"? 

No clue what that is unfortunately

usernametakenaaaa, 2022-05-30:
> This might be the same issue that I am having: #12345
>
> These trigger mine crashing: Attempting to open "Marketplace" Attempting to open "From Digital Factory" Attempting to open "Cura Backups\Manage [...]

One more location is the "multiply object", it will also crash Cura 5.

It's the good old multi-monitor bug. If you have more than one screen (for example a laptop with a docking station or just 2 screens) then cura will crash like described. No workarounds.

usernametakenaaaa, 2022-05-30:
You need to disable every monitor but one, Cura is not compatible with more than 1 screen connected.
It looks like the devs are seriously overloaded with work as that bug has been known since beta 1 release.
So only workaround is to remove the additional monitors,  then it will work.

usernametakenaaaa, 2022-05-30:
> Does this also happen when opening any of the other popup screens (Preferences, for instance?)

Preferences is not affected.
The splash screen and most smaller QT modals/dialogs are affected. They open on the wrong screen and in addition crash cura.

pbdroid, 2022-05-30:
As my old English teacher used to say, you are a gentleman and a scholar! Thank you for informing me of this bug. I tried with one monitor, and yes that seems to work.

I had originally talked to ultimaker and they did not offer this tip. It's a shame since it seems to be a known one. It would have saved me a lot of trouble and frustration. 

Even though Cura is a free product, I expected a little little customer service from them considering that I bought a S5 with material station and air manager this year, obviously a lot of money for one of their printers.

I hope that this is fixed soon though, as I always run dual monitors and really am annoyed to only be able to use one while working on this.

Vandresc, 2022-05-30:
Thank you for your feedback! Issue also brought in https://github.com/Ultimaker/Cura/issues/12102. 
We will discuss this again with the team.

voxpop9, 2022-05-30:
> So only workaround is to remove the additional monitors,  then it will work.

Can confirm this solves the issue. It's very unfortunate though as having to disable 2/3 of my monitors just for Cura 5.0 is not ideal.

Vandresc, 2022-05-31:
Devs See Cura-9344 for our internal reference.

usernametakenaaaa, 2022-06-25:
Half a year later .. Cura still crashes instantly.
After paying 5k on a printer I've to remove my monitors to print with it.

fieldOfView, 2022-06-25:
@usernametakenaaaa, though I can imagine this issue is highly annoying, it has not been half a year since may.

Until this issue is fixed, Cura 4.13 is still a very capable slicer and available here:
https://github.com/Ultimaker/Cura/releases/tag/4.13.1

pbdroid, 2022-06-27:
What I have the biggest problem with is Ultimaker not posting this "issue" on the main download page and then releasing this as a stable release with this major issue. It should still be a BETA with the warning. I spent a ton of time trying to figure this out, even talking to Ultimaker support, which said that they had no idea and told me to come here. I too bought a $10 machine (S5 with material station and filtering system) and had to get the help I needed from a fellow user.

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

- A surprising connection was found to some particular audio drivers: https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232
- Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

rburema, 2022-08-25:
@NeuralDynamics We hope it's related to the Qt bug, otherwise we don't have many leads. If it is indeed the Qt bug, there is a chance we can fix it on our own, as we're now looking into compiling our own (Py)Qt (which we can patch in time, instead of on their also under-personed schedule, because we think we might've found the bug).

That's a lot of _if_'s unfortunately. Best case is that we have it fixed in 5.2.x, but then _everything_ needs to go right in 1 go basically.

If all of that happens, we could put pointers to early builds in this thread to see if our proposed fix is working.

voxpop9, 2022-08-25:
@rburema Do you have an estimated date for this possible fix?

rburema, 2022-08-30:
@voxpop9 Unfortunately not. -- Other than the proposed dates for 5.2 that is, the Beta of which should be out by early October.

Golfern71, 2022-10-13:
Disable the Nahimic Service in Services like showed here: https://www.youtube.com/watch?v=XW-E1P6Ms4s
Works like a charm

cmp-nct, 2022-10-31:
I do not believe there is a chance they fix it, Cura devs do not have two screens to test it.
Cura 5 has been crashing since Beta.

What you can do on Windows 11 is install it on an emulator in WSL-2 subsystem with GUI.
This way you can emulate a single-monitor setup.

voxpop9, 2023-02-19:
So far this bug seems to be fixed for me although I do have a total of 3 monitors (2 identical, 1 tiny one for diagnostics)

voxpop9, 2023-02-19:
Nvm it has started crashing again

voxpop9, 2023-03-03:
Additional info in case this helps: it seems that the crashing problem does not happen at all until you try to port in old profiles from another version. Only then does it permanently start crashing on that specific version. e.g. When V5.2.1 gets my profiles from 4.13.1, it has the issue forever, even if I reinstall it. However, if I get V5.2.2 and I lost all my profiles from 4.13.1, there would be no issues and I can use it even with multiple screens.

FBruhns, 2023-05-15:
I can confirm that disabling "Nahimic Services", as proposed by @Golfern71  solved  my issues with dialogs freezing.