Ultimaker/Cura #12266: Cura 5 won't open files

VK1IAN, 2022-05-19:
### Application Version

5.0.0

### Platform

Ubuntu 20.04

### Printer

Cocoon Create

### Reproduction steps

1. Cura opens fine. I can change print settings and work my way through the menus. I can also configure Cura (Preferences->Configure Cura).
2. When I click the folder at top left (to open a file), nothing happens.
3. If I try it via File->Open File(s), nothing happens.
4. I can drag a STL onto the print bed, set up the parameters and slice, but then I can't save the resulting gcode to disk.
5. "Save to removable media" does work.

### Actual results

"Open File(s)" dialog box doesn't appear.
After I've dragged a STL onto the print bed and sliced, nothing happens when I click "Save to Disk" (i.e "save to disk" dialog box does not appear).

### Expected results

File dialog box should appear to allow me to open files or select the destination for a gcode file.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8734626/cura.log)

Vandresc, 2022-05-30:
Hi, @VK1IAN any chance you are using Debian 11?

VK1IAN, 2022-05-30:
> Hi, @VK1IAN any chance you are using Debian 11?

Ubuntu 20.04 (in a VirtualBox VM)

Vandresc, 2022-05-31:
Devs see CURA-9348 for our internal reference.

akshimassar, 2022-06-03:
The same thing happens for me. I'm using arch linux with AppImage

akshimassar, 2022-06-03:
That seems to be problem with opening "open" dialog. I successfully opened file from `Open Recent` menu.

akshimassar, 2022-06-03:
```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_cura5RySS4x/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_cura5RySS4x/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

akshimassar, 2022-06-03:
Corresponding action ( I guess) from Actions.qml
```
    Action
    {
        id: openAction
        property var fileProviderModel: CuraApplication.getFileProviderModel()

        text: catalog.i18nc("@action:inmenu menubar:file","&Open File(s)...")
        icon.name: "document-open"
        // Unassign the shortcut when there are more than one file providers, since then the file provider's shortcut is
        // enabled instead, and Ctrl+O is assigned to the local file provider
        shortcut: fileProviderModel.count == 1 ? StandardKey.Open : ""
    }
```

Is there anything I can do to help debugging?

akshimassar, 2022-06-03:
Save project file dialog won't open as well.

Claghorn, 2022-06-05:
Same problem on fedora 36. Running in a terminal shows this stuff:
`PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaVg8T4R/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaVg8T4R/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
qt.qml.context: file:///tmp/.mount_UltimaVg8T4R/share/cura/resources/qml/Settings/SettingView.qml:427:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
`

akshimassar, 2022-06-06:
Is there any workaroud for that issue? I would like to use cura 5 but for now I can't

minego, 2022-06-06:
You can drag a file onto cura to open it

akshimassar, 2022-06-06:
What about saving resulting g-code? None of the system dialogs of this kind work.
> You can drag a file onto cura to open it

minego, 2022-06-06:
I use octoprint so I haven't needed to save it

akshimassar, 2022-06-06:
Here is the error from corresponding cura.log file
```
2022-06-06 23:19:03,545 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///home/akshi/appimage/squashfs-root/share/cura/resources/qml/Cura.qml:629:5: QML FileDialog: Failed to load non-native FileDialog implementation:
qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml:37 module "Qt.labs.folderlistmodel" is not installed
```

akshimassar, 2022-06-06:
I've tried installing following packages `python-pyqt5 qt5-declarative qt5-quickcontrols` (to provide folderlist module) and the error from log is gone but open / save dialogs are still not working.

alexban011, 2022-06-06:
> What about saving resulting g-code?

save it to removable and then move it wherever you want. Annoying I know, hope they fix it soon. This is the reason why I'm still on 4.13.1 where I get like 8 error messages telling me the plugins fail

Goodfeat, 2022-06-07:
In windows 10 about the same problem, only writes an opening error
In general, I noticed that if the file is saved in binary format, then it opens, if left in ACSII, then the cura 5 refuses to open it

> 2022-06-07 12:48:03,397 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1751]: Attempting to read file file:///G:/Desktop/Кольцо.STL
> 2022-06-07 12:48:03,399 - INFO - [JobQueueWorker [7]] STLReader.STLReader.load_file [61]: Using legacy code to load STL data.
> 2022-06-07 12:48:03,400 - DEBUG - [JobQueueWorker [7]] STLReader.STLReader._read [102]: File did not contain valid data, unable to read.
> 2022-06-07 12:48:03,400 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]: Exception: Exception occurred while loading file G:/Desktop/Кольцо.STL
> 2022-06-07 12:48:03,401 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]: Traceback (most recent call last):
> 2022-06-07 12:48:03,401 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:   File "UM\FileHandler\ReadFileJob.py", line 78, in run
> 2022-06-07 12:48:03,401 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:     self.setResult(self._handler.readerRead(reader, self._filename))
> 2022-06-07 12:48:03,402 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:   File "UM\Mesh\MeshFileHandler.py", line 36, in readerRead
> 2022-06-07 12:48:03,402 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:     results = reader.read(file_name)
> 2022-06-07 12:48:03,402 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:   File "UM\Mesh\MeshReader.py", line 36, in read
> 2022-06-07 12:48:03,402 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]:     if mesh.source_mime_type is None:
> 2022-06-07 12:48:03,403 - ERROR - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [80]: AttributeError: 'NoneType' object has no attribute 'source_mime_type'
> 2022-06-07 12:48:03,403 - DEBUG - [JobQueueWorker [7]] UM.FileHandler.ReadFileJob.run [83]: Loading file took 0.0 seconds
> 2022-06-07 12:48:03,412 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=0), took 0ms

akshimassar, 2022-06-07:
Sorry for bothering, but how can we get updated version? I can only see 5.0.0 available at the main page

Claghorn, 2022-06-07:
At least I found a handy command line program on github named "dragon" which allows me to feed it a filename on the command line then drag that file into cura. About 10,000 times faster than bringing up an annoying GUI file manager and agonizingly navigating to the directory containing the file I want to drag. It does look like I can print directly to octoprint after slicing, so except when I want to edit gcode, I can probably use 5.0 now. Looks like I could download the gcode from octoprint, edit it, then re-upload as a round about way to do that as well.

minego, 2022-06-07:
Link? That sounds useful.

Claghorn, 2022-06-07:
dragon tool:

https://github.com/mwh/dragon/releases/tag/v1.2.0

rmsc, 2022-06-25:
> That won't work or matter. Cura 5.0.0 uses qt6, not qt5.

Is qt6 being statically linked? I didn't have qt6 installed, and installing all of it changed nothing.

rmsc, 2022-06-25:
BTW, a simple workaround is opening the file from the command line:

``` $ cura yourfile.stl ```

fieldOfView, 2022-06-26:
Qt6 is not statically linked, but the AppImage includes Qt6 (and all of the other dependencies) and uses that.

jellespijker, 2022-06-26:
Not opening the filedialog due to the folder list lib not being part of PyQt6-Qt6 was circumvented by us settings the
`QT_QPA_PLATFORMTHEME` in the AppRun script.

https://github.com/Ultimaker/Cura/issues/11874#issuecomment-1109757093

This seems to have mixed results. For my personal Laptop with Manjaro KDE I finally got the system filedialogs, @Ghostkeeper and @nallath reported quicker opening of their dialogs, but other Linux systems seems to break with this.

akshimassar, 2022-06-29:
Since you mentioned a line with `xdgdesktoportal` I was finally able to figure it out for myself.
Installing `xdg-desktop-portal` (`xdg-desktop-portal-gtk` implementation for me) fixed the issue for me. I never used flatpak and thus xdg desktop portal was never part of my system.
Note: I had to run `dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY` once after installing to prevent slow startup of applications and to make it work without restarting dbus.

I also tried some random values for `QT_QPA_PLATFORMTHEME` in AppRun file like `gtk2`, `qt5ct` and `qt6ct` but none worked for me except `xdgdesktopportal`.

ethanak, 2022-06-29:
Works for me! Thank you!
XUbuntu 20.04

qcol, 2022-07-07:
mee too, 5.1 beta - still can't open / save files

torrancew, 2022-07-13:
I'm also experiencing this issue on ArchLinux with the Cura5 AppImage.
Unlike @akshimassar, I have been unable to resolve it (I already have the XDG portal packages installed as I do use some flatpak-distributed things).

I think this problem may be related to the interaction AppImages have with the host system, I believe (reference: https://github.com/AppImage/AppImageKit/wiki/Similar-projects#general -- Row "`Using libraries and dependencies`", but have little experience with AppImages themselves, and am not sure how best to debug/chase that idea down.

Is there any plan/intention to eventually package Cura5 as a flatpak? If so, I'd gladly test my (quite possibly incorrect) theory out.

jellespijker, 2022-07-13:
@torrancew I think your issues with the AppImage relate to #11856  . 

Can you please try the `latest-Linux` AppImage from the link I posted at the end of that thread.
And to keep things a bit organized continue the discussion for that specific problem in that thread.

madnificent, 2022-07-14:
I have executed the following steps on Arch linux with the custom build mentioned in #11856 .  Based on https://github.com/Ultimaker/Cura/issues/12266#issuecomment-1170370529 of @akshimassar 

```
yay -S xdg-desktop-portal xdg-desktop-portal-gtk
dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY
```

This gives me the file open dialog through the UI.  It did not work when installing only `xdg-desktop-portal-gtk`.

Thanks for figuring out how to make it work on Arch Linux and kudos to the development team on answering the avalanche of questions that seem to pop up here.  People (including me) are really eager to test-drive Cura 5 XD

Ghostkeeper, 2022-07-14:
> Since you mentioned a line with xdgdesktoportal I was finally able to figure it out for myself.
Installing xdg-desktop-portal (xdg-desktop-portal-gtk implementation for me) fixed the issue for me. I never used flatpak and thus xdg desktop portal was never part of my system.
Note: I had to run dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY once after installing to prevent slow startup of applications and to make it work without restarting dbus.
> 
> I also tried some random values for QT_QPA_PLATFORMTHEME in AppRun file like gtk2, qt5ct and qt6ct but none worked for me except xdgdesktopportal. I'm using arch linux with awesome desktop manager.

Could I have some clarification on this workaround, @akshimassar ?
Currently in the application's AppRun script it already sets `export QT_QPA_PLATFORMTHEME=xdgdesktopportal`. So then the only change that you did for this workaround is to install that `xdg-desktop-portal` package?

I'm making some improvements to include the xdg-desktop-portal package in the build server so that it hopefully picks that along as a dependency inside of the application.

torrancew, 2022-07-14:
@Ghostkeeper The reported workarounds so far aren't solving the issue for me (Arch Linux/i3WM). I suspect there's still some undocumented piece here.

mwiercinski, 2022-07-14:
Still the case with 5.1.0-beta. 

This is crippling main user journeys with no workarounds and has been in the backlog for 2 months.

Ghostkeeper, 2022-07-15:
Note that a workaround is still to drag the file onto the main window. Anything else than crashing on opening the file dialogue (e.g. Cura crashing on start-up) should be seen as a separate issue.

I think there are multiple issues intersecting here, so I'm trying to fix one at a time.

Ghostkeeper, 2022-07-15:
If changing the `QT_QPA_PLATFORMTHEME` variable worked for one of you, could you please share the output of `echo ${XDG_CURRENT_DESKTOP}`?

rmsc, 2022-07-15:
> @Ghostkeeper The reported workarounds so far aren't solving the issue for me (Arch Linux/i3WM). I suspect there's still some undocumented piece here.

Is the `xdg-desktop-portal.service` user service started and enabled? At least in my case I had to do that after installing the required packages:

```
$ systemctl start --user xdg-desktop-portal
$ systemctl enable --user xdg-desktop-portal
```

torrancew, 2022-07-15:
@rmsc After a little digging, I was able to get this working. Some notes I took for @Ghostkeeper et al:
* I did not need to set `QT_QPA_PLATFORMTHEME` (it is completely unset in my shell/desktop environment)
* I did need to run `systemctl --user start xdg-desktop-portal.service` -- By default, this unit is not "enable-able" on ArchLinux (contains no `[Install]` section in the unit). Furthermore, xdg-desktop-portal was silently hanging for me, until I removed `xdg-desktop-portal-wlr`. I diagnosed this by running the desktop portal directly in a terminal via `/usr/lib/xdg-desktop-portal --replace --verbose`, and noticing some timeouts and a large delay before it would report `org.freedesktop.portal.Desktop acquired`. The system I'm using at the moment is an i3 system, but I had previously attempted an install of `sway`, which did not wind up working with my current NVidia card. The sway/wayland environment was largely still installed however, which likely caused this issue.

I'm going to look into filing a bug with xdg-desktop-portal upstream about what I encountered there. In the meantime, a concrete ask for the Ultimaker folks:
Since this systemd unit is configured (at least in Arch) as a dbus unit and provides no `[Install]` section, can Cura do something to trigger its launch, so that users don't have to manually start or patch this unit file?

torrancew, 2022-07-16:
After a bit more poking, I believe that this service is being auto-started (when installed) just fine, and that my previous issues with `xdg-desktop-portal-wlr` were masking that fact. @Ghostkeeper based on that, I believe that, _if xdg-desktop-portal_ is functioning correctly on a machine, adding that package to the build process should resolve this issue (and obviously, fixing broken xdg-desktop-portal installs for individual users is not Ultimaker's responsibility)

akshimassar, 2022-07-16:
> Could I have some clarification on this workaround, @akshimassar ?
I would be happy to help debugging. Although I have no experience with qt or dbus. Basically I installed `xdg-desktop-portal` and `xdg-desktop-portal-gtk`.

Arch package for xdg-desktop-portal depends on xdg-desktop-portal-impl which is virtual package and requires one of the following:
```
sudo pacman -S xdg-desktop-portal
resolving dependencies...
:: There are 5 providers available for xdg-desktop-portal-impl:
:: Repository extra
   1) xdg-desktop-portal-gnome  2) xdg-desktop-portal-gtk  3) xdg-desktop-portal-kde
:: Repository community
   4) xdg-desktop-portal-lxqt  5) xdg-desktop-portal-wlr
```

After that I think I ran `dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY` that came out somewhere in the slow startup bug reported to xdg-desktop-portal. I still have the problem with initial slow startup but since I rarely reboot it doesn't bother me.

Now I have both binaries `xdg-desktop-portal` and `xdg-desktop-portal-gtk` running under my user started about the time of last reboot.

chipmunk1977, 2022-07-19:
Not sure if this is a coincidence, or whether it has actually been fixed - but everything now works as expected after upgrading to 5.1 stable.  (However, my OS was also updated today to Ubuntu Studio 22.04LTS with the Plasma desktop.  If it hasn't been fixed, then maybe the OS update info will help.)

mwiercinski, 2022-07-22:
Installing xdg-desktop-portal{,-gtk} fixed the issue. Reproducible both on 5.0.0 and 5.1.0.

ethanak, 2022-07-23:
It is not real fix - Cura should suggest file name and folder for saving, but with xdg-desktop filename field is empty, and folder is rather random. Both Cura 5.0 and 5.1, XUbuntu 20.04.

jellespijker, 2022-07-23:
It is indeed not a fix, but a work around. We currently don't know how best to fix this bug and take into account all the different desktop and graphical managers.

We investigated shipping the xdg-desktop-*** in the AppImage but that is actually a system library and something like that is bad practice for a reason.

We plan on writing a support article to explain the current work around, to a broader audience then all you tech-savy people with a GitHub account.

We will ofcourse continue our investigation in a "real" fix for future release. But to temper the expectations a bit this will probably not be fixed for 5.2 except if we receive an external pull request. Which I sincerely hope to receive because my personal laptop also experience this issue.

Claghorn, 2022-07-23:
One "real" fix would just be to write a file browser in python (wouldn't be surprised to discover there already is one in the wild somewhere). Probably work better than the annoying bloated system browsers in any case :-).

dubiouser, 2022-07-24:
what was 4.x doing that was not failing, 
also:
  would it at least be possible to pop up an error instead of silently failing?

---
FWIW:
```
sudo apt-get install xdg-desktop-portal-gtk
dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY
```
is a functioning workaround for me - ubuntu 20,04 and cura 5.1.0

jellespijker, 2022-07-25:
> what was 4.x doing that was not failing

The 4.x builds and the 5.1 builds can't be compared. We had to upgrade from PyQt5 to PyQt6, apart from bumping up the minimum required libstdc version (Ubuntu 20.04 and higher) also had a change in the Qt library which are shipped in PyQt. We previously used our own file dialogs, instead of system dialogs. But PyQt6 is missing one of the required libraries for that (listdirs.so as far as I recall).

A possible solution could be to compile Qt6 from source, which is manageable in our current setup due to the fact that Qt now switched to CMake and Conan. However this will also require us to compile PyQt6 instead of just collecting the wheel files.

Speaking from experience compiling Qt and PyQt aren't trivial tasks, which also have a tendency to expand in scope very easily. Since we're still in the process of modernizing our current build / development environment. It is highly unlikely that we can introduce that extra work and risk at this point in time. Certainly if this possible solution isn't guaranteed to work.

mknj, 2022-07-25:
Ubuntu 22.04, i3 window manager.
I had to install xdg-desktop-portal-gnome and use the modern-linux appImage. (-gtk2/3 do not work)
Unfortunately the file dialogs are working suboptimal. If i try to enter the filename of an export, the cursor jumps to a search dialog after every letter. So im am still using 4.x for now.

jellespijker, 2022-07-25:
@mknj did you try installing `xdg-desktop-portal-gtk`? Maybe that solves your problem.

bullestock, 2022-08-05:
Can confirm the problem on Arch Linux, using Ultimaker-Cura-5.1.0-linux-modern.AppImage. Installing xdg-desktop-portal resolved the issue.

mknj, 2022-08-13:
`xdg-desktop-portal-gtk` is already installed at my system. and playing with the dbus activation did not help.
i guess my problem is related to using the i3 window manager. Cura 4 still works fine out of the box.

Systemblaster666, 2022-08-13:
Having the same problem using DWM. none of the suggested fixes work
I get this in the log when trying to open cura
` WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [442]: file:///(location)/Cura/squashfs-root/    share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
`

maxiauer, 2022-08-14:
Can confirm that problem exist also in ChromeOS 104 (Linux Container, Debian 11) which is a shame because lots of kids get this devices at school.

Also installing: 

sudo apt-get install xdg-desktop-portal-gtk xdg-desktop-portal

Works as a workaround, but the experience is not great. Extensions are not included by default and if for any reason the File window gets hidden by another window (if it gets "behind" another window), clicking on the Folder Icon does not bring it up to the front.

jellespijker, 2022-08-26:
We are currently trying to integrate the building PyQt6 and Qt6 from source in our build steps, which should open up some new possibilities to fix this problem in an uniform way. Because I think the root cause is that the PyQt6 wheel from the pypi repository doesn't contain certain binaries, such as `folderlistmodel` Once PyQt6 and Qt6 are integrated in our build process we can try fixing these issues.

FYI @MariMakes

fcuencamargalef, 2022-08-31:
The same with Ubuntu 22.04 + Cura 5.1.0 Modern

hpsaturn, 2022-09-10:
I have the same issue on Debian with this output:

```python
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_cura4nQTOs/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_cura4nQTOs/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

My setup:

```bash
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

zu4lu, 2022-09-21:
> I have executed the following steps on Arch linux with the custom build mentioned in #11856 . Based on [#12266 (comment)](https://github.com/Ultimaker/Cura/issues/12266#issuecomment-1170370529) of @akshimassar
> 
> ```
> yay -S xdg-desktop-portal xdg-desktop-portal-gtk
> dbus-update-activation-environment --systemd DBUS_SESSION_BUS_ADDRESS DISPLAY XAUTHORITY
> ```
> 
> This gives me the file open dialog through the UI. It did not work when installing only `xdg-desktop-portal-gtk`.
> 
> Thanks for figuring out how to make it work on Arch Linux and kudos to the development team on answering the avalanche of questions that seem to pop up here. People (including me) are really eager to test-drive Cura 5 XD

Thanks, this works even in debian. Modified the startscript and it is working now as expected.

LarrySikes, 2022-10-01:
Any update - my ticket is marked closed but issue still exists.

hpsaturn, 2022-10-03:
On my side, in **Debian 11** is working the last version _Ultimaker-Cura-5.1.1-linux_ via **AppImage**. Thanks guys.

svsdval, 2022-10-11:
> 

Debian GNU/Linux bookworm/sid, for Cura 5.1/5.2  did not help =( 
[cura5.1.log](https://github.com/Ultimaker/Cura/files/9758581/cura5.1.log)
[cura5.2.log](https://github.com/Ultimaker/Cura/files/9758582/cura5.2.log)

[42 more replies]

closed 2023-10-19 by MariMakes