Ultimaker/Cura #12268: Cannot open files using Cura on linux Chromebook

jasondilts, 2022-05-20:
### Application Version

5.0.0

### Platform

Linux

### Printer

Ender 3 Pro

### Reproduction steps

1. Downloaded Cura 5.0.0
2. Attempted to open a file to slice
3. Would not open when clicking on the folder icon
4. Would not open when clicking File, open File

### Actual results

Nothing happens, can't open a file to slice. Everything worked fine in the previous version of Cura

### Expected results

Expected to be able to open a file and slice it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

gleonard3, 2022-05-21:
This was also reported in Beta 5, Issue #11874

Vandresc, 2022-05-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/12266. 

Thank you for your feedback, we will bring this up with the rest of the team.

MariskaMaas, 2022-06-03:
Hey @jasondilts 

We don't support Chrome OS and unfortunately we are not planning developing Cura for Chromebooks any time soon.
But if your issue is indeed a duplicate of https://github.com/Ultimaker/Cura/issues/12266, than you can follow the progress there.

I will close this issue since it's a duplicate, I hope you understand.