Ultimaker/Cura #12269: Crashes while closing cura - VM Region Info: 0x17 is not in any region.  Bytes before following region: 4318330857

chuckthedudeman, 2022-05-20:
### Application Version

4.12

### Platform

MacOS 12.3.1

### Printer

Voron 2.4

### Reproduction steps

open a large stl file
slice
Preview
zoom in and out
change Z axis while viewing the preview

close application
wait for crash report to be displayed

### Actual results

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000017
Exception Codes:       0x0000000000000001, 0x0000000000000017
Exception Note:        EXC_CORPSE_NOTIFY

VM Region Info: 0x17 is not in any region.  Bytes before following region: 4318330857
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      101648000-101890000    [ 2336K] r-x/r-x SM=COW  ...ts/MacOS/cura
[cura.log](https://github.com/Ultimaker/Cura/files/8735428/cura.log)

### Expected results

clean exit without a crashed stack trace displayed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8735567/cura.log)

Vandresc, 2022-05-30:
Hi @chuckthedudeman, Is it possible for you to check if this still happens for you in Cura 5.0? 
Thank you!

Vandresc, 2022-06-03:
That is great to hear! We will close this ticket then, but feel free to re-open it if you encounter other related issues on your side. 
And thank you for trying Cura 5.0.

closed 2022-06-03 by Vandresc