Ultimaker/Cura #12270: UI > Overlay 'Material Estimation' > Information displayed overflows two a second line

Diegovd, 2022-05-20:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

1.- Open Cura 5.0.0  (or Cura 4.13 even, it was already present there)
2.- Select Printer "Ender-3 Pro"
3.- Open an STL File
4.- Slice
5.- On the print estimation information, below, right side: hover the mouse over the " i " (Info Icon) to display detailed information about the printing process (detailed time estimation for each line type + etimation called 'material estimation' (filament type, distance, weight, cost).

Precondition: 
- duplicate the "PLA" material
- assign curreny in the 'General' Settings chapter
- in the Materials chapter, under the selected material profile assign the 'Filament Cost' per kg. (in my case CHF and 25.-)

### Actual results

The 'material information' results overflows to a second line (actually the end of the cost value overflows in his cell to a second line).

Please see the screenshot in the attachments.

### Expected results

The 'material information' results are displayed on one line only.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

If further information is needed, plese let me know.

Thank you very much.
Best regards, Diego von Deschwanden.

Vandresc, 2022-06-03:
Hi @Diegovd , that you for your feedback. We discussed this case and unfortunately due to all the other high prio issues on our backlog we decided to defer this ticket, meaning it will still stay open but we don't have it on our roadmap, just yet. I hope you understand.

Diegovd, 2023-06-03:
Hello,
The bug is still present in version 5.3.1
It is not a problem for me, it's just a bug, it can be solved easily I think,

I took some time to look where it could be configured. I think I found it:

Cura 5.3.1:
Cura-main\Cura-main\resources\qml\ActionPanel\
File: PrintJobInformation.qml

Line  120:             property var columnWidthMultipliers: [ 0.46, 0.18, 0.18, 0.18 ]

The last "0.18" seems to be for the cost. It's too small.
Note: it seems that adding all the values in brakets needs to equal 1.

If you decreased "0.46" to "0.36" and add 0.10 to the last "0.18", then "0.28" being bigger, maybe it would not overflow to a second line.
I have no means to test that.

I would try:

Line  120:             property var columnWidthMultipliers: [ **0.36**, 0.18, 0.18, **0.28** ]

[image]

[image]

Thank you :)

Diegovd, 2023-06-04:
OK, so I was able to modify the file locally and test it.

C:\Program Files\UltiMaker Cura 5.3.1\share\cura\resources\qml\ActionPanel\PrintJobInformation.qml
Line 120: property var columnWidthMultipliers: [ **0.36**, 0.18, 0.18, **0.28** ]

And the result:

[image]

MariMakes, 2023-06-05:
Hey @Diegovd,

Nice to meet you 👋 I'm the fresh set of eyes that started the bot.
Thank you so much for investing extra time in this and putting effort into finding a solution. 😍 🎉 

I'll bring your issue up to the team to see if the fix is generic enough to be included in the next Cura release. 💪

Diegovd, 2023-06-05:
Hello @MariMakes, thanks for your reply.
To be honest: I did further testing with an oversized object to print. My solution is OK for the Cost part, but the filament lenght in meters will also overflow to a second row if the number of characters is too big.

[image]

That is the problem of having fixed width multiplier values to set the colums width of each text&value.
I did a little bit of research into qml, but it's not something that I have used before.

So I tried these values:

Line 120:             property var columnWidthMultipliers: [ **0.26**, **0.28**, 0.18, **0.28** ]

That seems to be good.

[image]

If there is anything I can do to help, please let me know.

Thank you.

Regards, Diego

MariMakes, 2023-06-28:
Quick update from our side 👋
We really appreciate the effort you put into finding a solution. 
I transformed your super clear proposal into a pull request, hopefully we can include it in the next Stable release. 

Thanks again 😊

Diegovd, 2023-07-09:
Hello @MariMakes,
I apologize for this late reply. Thank you very much for your help!
Also many thanks for the mention as contributor. That made me feel very proud :)

I installed 5.4. Works perfect.
I will close this Issue now.
Thank you and best regards, Diego.

closed 2023-07-09 by Diegovd