Ultimaker/Cura #12272: Scrolling through any list in Cura is not accurate following the mouse-wheel

MIOsystems, 2022-05-20:
### Application Version

5.0

### Platform

Microsoft Windows [Version 10.0.17134.677]

### Printer

UM3

### Reproduction steps

Scroll trough any list in Cura will frustrate you to the max because it keeps scrolling although you are not scrolling with your mouse-wheel anymore.

See screen recording for a good experience on this issue.

### Actual results

List-scroll stops at a random moment. 

### Expected results

Just follow the mouse wheel as any other program does in windows.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/46918729/169497464-3d2428ed-fc46-4259-be0f-ce66dd468b81.mp4

Triangulix, 2022-05-28:
I experience the same issue. Cura is enforcing "smooth scrolling", a "feature" that normally can be turned off in Windows system settings if a program doesn't enforce it.

Vandresc, 2022-05-31:
Thank you both for your feedback. We have a similar ticket on our backlog, devs see CURA-9164 for our internal reference.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant