Ultimaker/Cura #12276: Raft does not print completely

yanshengk, 2022-05-20:
### Application Version

5.0.0

### Platform

macOS Monterey 12.4

### Printer

Custom FFF Printer

### Reproduction steps

1. Build Plate Adhesion Type changed to Raft. 
2. Raft Extra Margin changed to 10.0 mm. 
3. Raft Air Gap changed to 0.14 mm. 
4. Sliced the model. 

### Actual results

The third layer of the raft did not print completely. 

Layer 2
[image]

Layer 3
[image]

### Expected results

The third layer should be printed completely covering the whole layer 2. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file and screenshots of the bug were included in the ZIP file.
[Archive.zip](https://github.com/Ultimaker/Cura/files/8740436/Archive.zip)

Vandresc, 2022-05-30:
Hi @yanshengk thank you for your feedback, I could indeed reproduce the issue with your project file. Will bring this up with the rest of the team.

Vandresc, 2022-05-31:
Devs, see CURA-9349 for our internal reference.

yanshengk, 2022-06-02:
> Hi @yanshengk thank you for your feedback, I could indeed reproduce the issue with your project file. Will bring this up with the rest of the team.

Thank you.

GregValiant, 2023-01-09:
Hi @MariMakes 
This is a follow up to your request in #13353 
Here is the project file sliced as I received it.  This is showing layer 3 with the model at X0 Y0
[image]

Here is layer 3 when moved to X-30 Y0
[image]

snebragd, 2023-02-05:
Hej! I ran into this very same problem today. Quite annoying as the top raft layer was hiding the problem in the preview. I only noticed mid print. Cura 5.2.2, MacOs 13.1

[image]

MariMakes, 2023-08-25:
Quick update from our side 👋

Good News! We are cleaning up our open issues and stumbled upon your issue about the raft top layer. We made some improvements in the buildplate adhesion and it seems like we cannot reproduce your issue in the upcoming 5.5 which means that the issue is fixed!

[image]

I'll be closing this issue, and you can use the fix in the upcoming 5.5 release.

closed 2023-08-25 by MariMakes