Ultimaker/Cura #12278: Cura 5 Hard Crashing

wolfbladet, 2022-05-20:
### Application Version

5.0-Beta1

### Platform

Windows 10

### Printer

Creality CR-10s

### Reproduction steps

I seem to do simple tasks, e.g. moving objects around, or after slicing previewing layers, and Cura crashes hard. 

### Actual results

Cura freezes and or crashes my machine completely, causing it to close Cura.

### Expected results

Simply not crash when moving objects / scrolling up and down in layer previews

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8742434/cura.log)

wolfbladet, 2022-05-20:
Seems like when I just replicated it again, this is the last entry in the log, while cura is hung / crashed:
2022-05-20 13:44:16,545 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [162]: Synced cloud printers with account.

wolfbladet, 2022-05-20:
I tried disabling the UM3 network plugin, with no change.

nallath, 2022-05-23:
Did you also try it with the 5.0.0 release (not the beta)?

wolfbladet, 2022-05-27:
I didn't realize that it wouldn't prompt to update, but yes, I just updated
it manually, and see the same.  The last entry in my log when Cura hangs is
always the autosaving message:

2022-05-27 11:19:18,335 - DEBUG - [MainThread]
UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode
object: 'four_square_cons.stl'>,trans.=<0.000,0.000,20.994>), took 1ms
2022-05-27 11:19:20,037 - DEBUG - [MainThread] cura.AutoSave._onTimeout
[64]: Autosaving preferences, instances and profiles took
0.012956857681274414 seconds

BulletDropplay, 2022-05-28:
> ### Application Version
> 5.0-Beta1
>
> ### Platform
> Windows 10
>
> ### Printer
> Creality CR-10s
>
> ### Reproduction steps
> I seem to do simple tasks, e.g. moving objects around, or after slicing previewing [...]

What is your system? CPU MoBo RAM GPU
seems like its the same issue as i was about to report yet again,
i used to have a ryzen 2700x and a gtx 970 but i upgraded cpu and gpu and are running a ryzen 9 5900x and a rtx 3089Ti but i kept my MoBo a x470 ASUS one and as far back as cura 3.4 i think it was ive had BSOD issues never encountered in anything but cura, my pc crashes and restarts randomly after 3-15min in cura, after i upgraded to cura 5 it seems it takes a bit more time, 5-15 min

nallath, 2022-05-31:
Well, if it's ryzen, it might be a known issue (and unfortunately, one we can't fix!)

tlhintoq, 2023-02-23:
24feb2023 - Not quite a year later - Same behavior as original poster on latest 5.3.0-beta.1

I've always made a point of being a little slow at certain things to give Cura time to 'catch up' and that actually works as silly as it sounds.

- Slice, preview, then wait a bit.  
- Don't have screen scaled other than 100%
- Don't have custom cursor arrow or mouse trails
- Don't mouse over it fast and speedy in preview mode.
- Don't race through the layers slider in preview mode.
- Don't move or rotate the view in preview mode if you can avoid it
- Don't move objects or blockers with haste.
- Drag slow
- Move slow
- Reposition slow

Just give it a couple seconds between actions to... I don't know... pre-cache what it expects it will need next?
And honestly, strangely, that's what keeps Cura viable for me.

But... Trying this 5.3.0-beta... It is definitely MORE FRAGILE for locking up.

- Launch
- Add Benchy... No slicing yet.
- Try to move Benchy... and locked.

Happens on all monitors.

[image]

tlhintoq, 2023-02-23:
[image]

Every installed version of Cura on my PC are set to use the High Performance GPU.  So I can't fathom how this PC can't handle the load.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant