Ultimaker/Cura #12279: OMG in MacOS Monterey it doesn't work at all

jma-grafik, 2022-05-20:
### Application Version

5.0.0

### Platform

MacOS Monterey

### Printer

CR-10 V3

### Reproduction steps

-If you just right click to move around the object, it will stick right-click forever. The only option is to Command-Q
-You can't click anywhere, nothing happens if you click to change settings, the drop menu will not appear
-the left side tool bar, the same, can't click anything. Sometimes, and just sometimes, if I go to Finder and then come back to Cura, it will allow me randomly to click in the first place I click, and then, will not be allowed to click in another menu, tool bar or whatever
-randomly loads the object dropped from the Desktop, other times will freeze the Cura app, and then force quit

Well, I almost could not make anything, just to open the app, and see that anything was working as expected, and give up

### Actual results

What? sorry, can't understand this… I already wrote above what happens with the new 5.0.0 version, that's all

### Expected results

the problem is just the user interface not working at all

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

the problem is just the user interface not working at all

GT-Geo, 2022-05-20:
I'm running Monterey also and have not experienced any of the issues you've explained.

jma-grafik, 2022-05-21:
> 

Yes, since I moved by force from Mojave I'm having a tough electronic life… I know most of the troubles are due to Monterey and/or eGPU. Thanks for making me know that other people are not having issues. Maybe I will delete all previous installations and download it again, and see what happens.

jma-grafik, 2022-05-21:
OK, I think I discovered the problem: Can't use the pen tablet because can't click correctly. I must to say that the pen tablet works perfectly with all the other applications

3dHawwy, 2022-06-09:
@jma-grafik, I have exactly the same problem.  Also Mac Monterey and trying to use my Wacom tablet.  It works on everything else without issues.  I tried another Wacom (from my MacBook)....same problem.  Cura 5.0 does not like tablets.  I try using my trackpad, but I don't like it.  And I never use a mouse, so I don't like that either...