Ultimaker/Cura #12283: Cura 5.0 - Text missing, hard to read

jwhite, 2022-05-20:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Voron

### Reproduction steps

1) Launch Cura
2) Read Text

### Actual results

[image]
[image]

### Expected results

Text should be clear like in Cura 4.x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

Vandresc, 2022-05-31:
Hi @jwhite, will bring this up with the rest of the team, thank you for your feedback.

Vandresc, 2022-06-03:
Devs, see CURA-9364 for our internal reference.

carribeiro, 2022-07-06:
I had opened #12630 which was deemed a duplicate of this one. I'd like to add that in that case, the rendering issue was not a purely aesthetical one as it affected the display of a value (horizontal expansion), which was misleading and led me to think that the correct value was not being applied (it was).