Ultimaker/Cura #12284: Merging models merges the model from its initial state without transformations

rhoogenboom, 2022-05-20:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 3 ext

### Reproduction steps

1. Load two separate STL models

[image]

2. Resize and/or rotate one or more of the models

[image]

3. Merge both models

[image]

### Actual results

The merged result is merged as if no rotation or scaling occured. They are merged with the state of the initial import. 

[image]

### Expected results

The models should be merged using the applied transformations, like rotation and/or scaling applied to each STL. This happens in cura 5.0.0.

In Cura 4.3.0 this works correctly, see screen print.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional information, issue seems to be persistent

marciot, 2022-05-27:
It has always been my understanding the "Merge" should treat both STLs as having a common coordinate system, and hence transformations need to be discarded, while "Group" will allow both STLs to maintain transformations and non-overlapping coordinate systems.

nallath, 2022-05-30:
@marciot Is 100% correct!

rhoogenboom, 2022-05-30:
> Speaking as a Cura user, it has always been my understanding the "Merge" should treat both STLs as having a common coordinate system (for dual printing where each STL specifies a different material within a common model), and hence transformations need to be discarded, while "Group" will allow both STLs to maintain transformations and non-overlapping coordinate systems. If my understanding is correct, then this would not be a bug, but the correct behavior.

Hi,

I agree that should start with the same coordinate system. But why you would reset the transformations?! If you didn’t apply any, it would work fine like you suggest it should work by default. But if you have to apply transformations, you’re basically stuck with an unprintable solution. I get the dual head bit, in such a situation you always have two models which were drawn as one. 

But I print a lot of PVA also on my UM and the shadow object used for the support is never from the same drawing. So basically, if if I have to print support now, I cannot resize it to the best suitable shape anymore because you delete the transformations. 

That would put a big ‘unusable’ label on using Cura 5 with my Ultimaker, which is a bit of a shame as it seems to slice faster.

rhoogenboom, 2022-05-30:
> @marciot Is 100% correct!

That might be 100% correct, but it’s also like 75% functional and definitely 0% backwards compatible with 4.3. 

As you can see above in the samples. With 4.3 I’m able to compose a perfect printable solution for my Ultimaker. With 5, I cannot print it. I could go back to the client, ask them to redo the other objects too so I need new STLs from things they drew one two years ago. Again for a dual object material print, probably in a few cases it would be a problem. 

But it would make my PVA print completely useless, as I wouldn’t be able to resize the PVA object to best support the main object. 

Anyway, it what it is I guess.

nallath, 2022-05-30:
A work around is that you can select the object inside a group with ctrl click and then transform them. 

But the merge solution was always intended for models that were designed with the same origin. I do remember that it used to be different, but it was super tricky (if not outright impossible) to handle all cases. I think you're actually the first person that ran into the simpler behavior since we changed it.

rhoogenboom, 2022-05-30:
> 

Well, I can see what I can get to with merges. Have to pay close attention to the transformation then. As I have to ensure they ‘same axis’ starting point. 

Geez, didn’t think I was the first. I would have thought lots of people use that as it is so powerful. But then again, maybe dual printing and complex transformations will only be applicable to a few. 

I must say I use the dual material option with transformations a lot for PVA removable support printing. Setting a second model as PVA gives you far more control and better base then just a support option. 

Cheers for the workaround!

marciot, 2022-05-30:
> Cheers for the workaround!

@rhoogenboom: I don't understand why it is a workaround. If I understand your use case correctly, using "Group Models" instead of "Merge Models" does exactly what you need, only it has a different name. In previous versions of Cura, "Group Models" and "Merge Models" did the same thing, which was redundant.

rhoogenboom, 2022-05-30:
> @rhoogenboom: I don't understand why it is a workaround. If I understand your use case correctly, using "Group Models" instead of "Merge Models" does exactly what you need, only it has a different [...]

It’s a workaround as it does not do the exact aame thing. If it did, I would have the same problem I have with the merging feature. 

Merging always takes the models from there common X Y and Z offset. Grouping takes the models from where-ever they are on the build plate. As you can see in the sample above, I have 34 different categories to print, so the alignment is pretty important otherwise the text is all over the place and you easily see the difference between different prints. That’s now manual work for me to make sure I align all 34 exactly the same in cura 5 which merging did automatically for me in cura 4.3. 

That’s workaround and thus not the same thing.

marciot, 2022-05-30:
@rhoogenboom: I see. It looks like you were taking advantage of the fact that "Merge" got rid of some transformations (the translation) while keeping others (the rotation). While this made little sense, for your particular STL files, it happened to work well.

With the new behavior, you have to manually do these steps, which is pretty hard if you want to do things repeatedly. A better way would be if you could use a mesh editor to bake in the rotation into the STL files for the base and the date STL so that they compose correctly using Merge. I'm not sure where those original files came from, but if you have a way to edit them, that would be the best solution, as it would allow you to do things quickly again.