Ultimaker/Cura #12285: Material Manager hangs/fails when duplicating Generic PETG setting

LilBub, 2022-05-21:
### Application Version

5

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Enter Material Manager
2. Select Generic PETG
3. Hit the 3 bars above and select Duplicate
4. Long hang
5. Enter in Display Name, Brand, Material type
6. Color, click Material Color Picker and select a color.
7. Long hang, then the previously filled out fields turn grey and you're unable to edit any more fields/print settings.
8. You ponder life choices
9. You wonder...maybe Simplify3D isn't so bad after all.
10. Then you are tempted to kick your cat.  But why blame the feline?
11. You exit the Preferences Menu (upper right X)
12. Top of screen, click the material, select down arrow and back into Material Manager
13. Then you can get back to work finishing the fields needed (i.e. type "green" in the Color text box)
14.

### Actual results

Covered above

### Expected results

Covered above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Vandresc, 2022-05-30:
Hi @LilBub, thank you for your feedback. We agree that your cat shouldn't suffer, we are all cat lovers in the Cura team, so let's see how can we avoid that ;) 

First of all, I couldn't reproduce the hanging when duplicating material, though I do agree it can be somehow slow but not seen on all machines. Will bring this up with the team. 
Second, I did experience the grey fields when duplicating the PETG (and not only), this happens when you will change the Brand type. Because the brand is not Default generic, it will of course add a new brand/category material down the list, containing the given name. This is all expected, what is not expected is that when the brand type is changed the duplicate material is somehow not selected anymore, this can be confusing, will bring it up with our UX designer. Keep you posted

MariMakes, 2023-05-15:
Hey @LilBub,

Quick update on our side.
We might have a fix for this issue. 🎉
It's a bit hard for us to test, but we have a nightly build available.

Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

LilBub, 2023-05-15:
> Hey @LilBub,
> 
> Quick update on our side. We might have a fix for this issue. 🎉 It's a bit hard for us to test, but we have a nightly build available.
> 
> Can you please try it out and let us know if it resolves your issue? You can find the download links and the discussion here: #15098

Yes I can do that.

I hope this fixes it, I get a lot of Microsoft spinning circles when duplicating and editing a material profile