Ultimaker/Cura #12286: No selection highlighting in Machine Settings Start and End GCode edit boxes

gordon101, 2022-05-21:
### Application Version

5.0

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

Steps to reproduce:

1. Open Machine Settings for any printer
2. Select existing Start G Code or End G Code. There is no selection highlight, although the selection is valid and can be Deleted, Copied or Pasted into. Just need to add selection highlighting, everything else works fine. 

### Actual results

The text editing selection works but it is invisible. Need to turn on selection highighting

### Expected results

When you select text in either Start G Code or End G Code it should be highlighted so you can see what is selected.

This is purely a UI issue so I don't think log file will help much

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8746227/cura.log)

Vandresc, 2022-05-31:
Hi @gordon101 will bring this up with the rest of the team. Thank you for your feedback!

Ghostkeeper, 2022-06-03:
We could all see the selection fine, like this:
[image]

Could it be that you have a system theme that changes the colour highlight of selected text to a very light grey colour that happens to (almost) match the background colour of that text field?

I see that it's not themed like other text fields are. I'll fix that issue.

CreeT6, 2022-06-17:
I have same issue on windows 10

MariMakes, 2022-08-22:
Hey all, This issue is resolved in Cura 5.1, see the commit from @Ghostkeeper.
I'm closing this issue. 😄

closed 2022-08-22 by MariMakes