Ultimaker/Cura #12287: Unable to fix gaps around small holes, because optimize wall print order ignores extra skin walls.

aldencolerain, 2022-05-21:
### Application Version

4.13.1

### Platform

Linux

### Printer

Monoprice Select Mini V2

### Reproduction steps

Slice and print any shape with a small hole in the xy plane.

### Actual results

Prints frequently have a thin gap between walls around small circular holes.  I think this issue is likely due to shrinkage between walls.  Optimize wall print order should help mitigate this, however this settings seems to ignore extra skin walls.  From the conversation below, I believe people have had similar issues for quite some time that seem to be resolved when using a different slicer.

https://community.ultimaker.com/topic/19312-thin-gap-between-outer-wall-and-inner-wall-only-on-small-circular-insets-but-not-on-the-perimeter-of-the-print/#comments

Cura without optimize wall print order:
[image]

Cura WITH optimize wall print order (notice yellow extra skin wall):
[image]

Prusa:
[image]

### Expected results

I expect Cura to print matching inner, outer, and extra skin walls one right after another, without a large time delay between them.  It looks like Prusa slicer does not have this issue.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[corner_clip.zip](https://github.com/Ultimaker/Cura/files/8746908/corner_clip.zip)

Vandresc, 2022-05-31:
Hi @aldencolerain , this should be already fixed in the 5.0 version. Let us know if this is fixed for you as well. Thank you!

aldencolerain, 2022-06-04:
Thank you @Vandresc.  I upgraded and it does look improved in version 5.  It looks like there are still times where it comes back to the circle for the skin, but other times it certainly does the skin immediately.  I don't know if there is some trade off.  I will close for now and investigate a little further. Attached are the two images for future reference if the issue persists for people
[image]
[image]
.

aldencolerain, 2022-06-12:
Thank you @Vandresc.  I upgraded and it does look improved in version 5.  It looks like there are still times where it comes back to the circle for the skin, but other times it certainly does the skin immediately.  I don't know if there is some trade off.  I will close for now and investigate a little further. Attached are the two images for future reference if the issue persists for people.

aldencolerain, 2022-11-02:
Looks like this is still an issue for me.  Here is another image from Cura 5.1.0 with optimize wall order on.  I still get gaps around holes I think because of cooling and the time between walls.
[image]

Vandresc, 2022-11-03:
Hi @aldencolerain, could you please share a project file? Thank you!

BogusF, 2022-11-20:
I believe I've got the same issue or at least its's something very similar. I'm using Cura 5.1, printer is an Ender 3 S1 Pro running Marlin based [mriscoc firmware](https://github.com/mriscoc/Ender3V2S1/releases/tag/20221002). Here's a project file:

[CE3_Abflusssieb.zip](https://github.com/Ultimaker/Cura/files/10050509/CE3_Abflusssieb.zip)

The underextrusion happens mostly in the topmost part of the sieve. At the bottom it's also there, but it's not nearly as bad. On the top, it basically prints a single wall hollow cylinder into thin air. 

In this print, I tried to remedy the underextrusion with "Retraction extra prime amount", that only brought me some stringing, but the underextrusion stayed, so I turned it off again.
[image]
[image]