Ultimaker/Cura #12289: Couches du radeau

alaintrt, 2022-05-21:
### Application Version

5.0

### Platform

Windows 10

### Printer

Geetech A20

### Reproduction steps

Le radeau ne s'imprime pas correctement en 3 couches complètes et successives. La couche 1  se fait en diagonale. La couche 2 se fait aussi en diagonale , mais partiellement, et la couche 3 couvre la totalité de la surface du radeau y compris les zones ou la couche 2 n'a pas été imprimée. 
En PJ la pièce à imprimer et les paramètres radeau.
Merci de votre aide
Cordialement
 AT  
[module 110 ouvert 50X70.zip](https://github.com/Ultimaker/Cura/files/8747227/module.110.ouvert.50X70.zip)
 

### Actual results

Mauvaise adhérence du radeau Radeau incomplet.

### Expected results

Le radeau devrait être comme dans la version 4.13.1 en 3 couches ou plus entrecroisées. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Rien à ajoute
[module 110 ouvert 50X70.zip](https://github.com/Ultimaker/Cura/files/8747237/module.110.ouvert.50X70.zip)
r

nallath, 2022-05-23:
Please fill in bug reports in English, we can't handle them otherwise.

alaintrt, 2022-05-23:
The raft does not print correctly in 3 full and successive layers. Layer 1 is done diagonally. Layer 2 is also diagonal, but partially, and Layer 3 covers the entire surface of the raft including the areas where Layer 2 was not printed.
Attached: part to be printed and the settings for raft.
Thank you for your help
Faithfully 
AT

Results:
 Poor adherence of the raft to the bed. Incomplete raft. Some part to be printed are missing. 

Expected results 

The raft should be as in version 4.13.1,  in 3 or more interlaced complete layers. 

[module 110 ouvert 50X70.zip](https://github.com/Ultimaker/Cura/files/8756692/module.110.ouvert.50X70.zip)

alaintrt, 2022-05-23:
nothing to be added as comments

closed 2022-05-23 by alaintrt