Ultimaker/Cura #12290: Part Cooling Fans Not Coming On

Wakinyantanka, 2022-05-21:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom BLV Cube

### Reproduction steps

Made sure to enable part cooling fans at desired layer height. They still wont turn on. 

### Actual results

Nothing

### Expected results

Part cooling fans should turn on.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[general]
version = 4
name = custom_extruder_1 #2_user
definition = custom

[metadata]
type = user
setting_version = 20
extruder = custom_extruder_1 #2

[values]
cool_fan_speed = 50.0
cool_fan_speed_0 = 5
cool_fan_speed_max = 100.0
inset_direction = inside_out
material_flow = 84.0
material_flow_layer_0 = 82.0
material_print_temperature = 240
meshfix_extensive_stitching = True
optimize_wall_printing_order = True
retraction_amount = 3
retraction_speed = 50.0
roofing_layer_count = 1
skirt_line_count = 4
speed_print = 80.0
wall_0_inset = 0.2
wall_thickness = 1.2

GregValiant, 2022-05-21:
In your gcode files there should be M106 lines that tell the fan to turn on (Example: M106 S255 would be fan at 100%).  It looks like your Initial Fan Speed is at 5% (M106 S13).  There are very few fans that will start to turn at a speed that low.
If you zip a gcode file and post it here it would be helpful.

Wakinyantanka, 2022-05-21:
It doesn't matter where I set the speed the fans will not come on.
[CFFFP_calibrationbox2.zip](https://github.com/Ultimaker/Cura/files/8748197/CFFFP_calibrationbox2.zip)

GregValiant, 2022-05-22:
If that is a single extruder printer then go into Machine Settings.  On the Extruder tab is a box for the fan number.  It should be set to "0".  It's the fan number, not the number of fans.
M106 S255 P1 is trying to start Fan #1 and your cooling fan is likely wired as Fan #0.

Wakinyantanka, 2022-05-22:
That's what was wrong. Totally my mistake, new to Cura. THANKS!!!!

closed 2022-05-22 by Wakinyantanka