Ultimaker/Cura #12291: Cura 5.0 on ubuntu 22.04 doesn't allow to login to my account

petervflocke, 2022-05-21:
### Application Version

5.0

### Platform

Ubuntu 22.04

### Printer

geeetech A10

### Reproduction steps

1. New installation
2. Run application
3. Click on sign-in in the upper right corner
4. And clik on the "sign in" button
5. Nothing happens

### Actual results

```
peter@alfa:~/Development/3DPrinter/cura$ ./Ultimaker-Cura-5.0.0-linux.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaD8PcMD/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaD8PcMD/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```
Here the second "sign in" button was pressed:

```
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_tls_channel_binding_error_quark
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
mkdir: /tmp/.mount_UltimaD8PcMD/libselinux.so.1: no version information available (required by mkdir)
/usr/bin/google-chrome-stable: symbol lookup error: /lib/x86_64-linux-gnu/libpango-1.0.so.0: undefined symbol: g_memdup2

```

Browser does not open login page

### Expected results

Brwoser open a login page and I can sign in 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura version 4.8 works fine.

Vandresc, 2022-06-03:
CURA-9342, devs see ticket for our internal reference.

firefloc, 2022-06-04:
I can give this log if this help pop 22.04
cura 5.0 appimage
only action is clicking on the signin button

`./Ultimaker-Cura-5.0.0-linux_07ecc3f54905e865167d2bcd7cfe459c.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaBwyqJ6/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaBwyqJ6/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_tls_channel_binding_error_quark
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
mkdir: /tmp/.mount_UltimaBwyqJ6/libselinux.so.1: no version information available (required by mkdir)
/usr/bin/google-chrome-stable: symbol lookup error: /lib/x86_64-linux-gnu/libpango-1.0.so.0: undefined symbol: g_memdup2

`

herrnici, 2022-06-06:
**Workaround:** Clicking the button in an older version of Cura will let you stay logged in in Cura 5.0.

Of course, having to have multiple versions of the same software installed isn't a long-term solution.

petervflocke, 2022-06-06:
> Clicking the button in an older version of Cura will let you stay logged in in Cura 5.0.

... not on ubuntu, since cura saves its profile in version depended folder: ~/.config/cura/"version folder" (4.11, 4.13, 4.8, 5.0)/cura.cfg

What one can do, is to start the last working version (for me 4.8), log in, close the app, and copy from the /old_version/cura.cfg file "ultimaker_auth_data" entry to the cura 5.0 config file.

It works for me and now I can work with v5.0 with signed int status. The login will eventually expire, but then I will repeat the above process. Still hopping that the issue will be fixed by cura guys soon.

@cura, please don't include the version in the auth data :)

nallath, 2022-06-08:
> please don't include the version in the auth data :)

We don't ;)

MariMakes, 2023-10-19:
👋 Quick update on our side. 

This should be resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

PR #12736 (2022-07-12): Cura 9205 appimage not launching on ubuntu 2204
Building two AppImages, one based on Ubuntu 20.04 and one with on Ubuntu 22.04. This also solves the sign in button not working on some Linux systems. Keep in mind that this is more of an intermittent solution and that we should still try to solves this in a more generic/future proof-way

Fixes both CURA-9342 and CURA-9205

The results of this run can be found here: https://github.com/Ultimaker/Cura/actions/runs/2648505664

- Fixes #11856
- Fixes #12453
- Fixes #11900
- Fixes #12232
- Fixes #12291
closed 2023-10-19 by MariMakes