Ultimaker/Cura #12293: Regression: monitor screen not fully shown on low resolution

Sewef, 2022-05-21:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Get 1360x768 resolution
2. Run a print
3. Profit

### Actual results

Right panel has a scrollbar.

[image]

### Expected results

Cura 4 didn't have this scrollbar.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

/

Vandresc, 2022-05-31:
Hi @Sewef , thank you for your feedback. By using the scrollbar are you able to scroll down and see the other information on the monitor page?

Sewef, 2022-06-10:
Whoops, forgot to answer
Yeah I can scroll to see the other informations. The problem is I didn't need to scroll back in Cura4 and it's not conveniant when I want/need to monitor everything at once