Ultimaker/Cura #12294: Interaction between 'extra skin wall count' & ' Skin overlap percentage' creates errors in skin  overlap

PhilBaz, 2022-05-21:
### Application Version

5.0

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Set 'extra skin wall count' = 1 & ' Skin overlap percentage' = 100.  Skin overlap % seems to have no effect.

Set 'extra skin wall count' = 0 & various settings for ' Skin overlap percentage' = 100.  Skin overlap % is correct.

Also, while playing with these setting along with skin pattern I got weird results where the skin-walls would start to overlap, rather than just the skin over the skin-walls. (ill include an image)

### Actual results

above....

### Expected results

Skin should overlap as requested...… Also, bridging needs its own skin overlap setting...please, please do this. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

Vandresc, 2022-05-31:
Hi @PhilBaz , thank you for your feedback. Could you please share the project file which reproduces the issue with us? Thank you!

PhilBaz, 2022-05-31:
Play with the 'extra skin wall count' & ' Skin overlap percentage' parameters and you'll see different types of weirdness.

Just want to again campaign for separate 'Skin overlap percentage' for bridging....please

[Untitled.zip](https://github.com/Ultimaker/Cura/files/8807395/Untitled.zip)

nallath, 2022-06-14:
We've added this ticket to our backlog. For future reference: CURA-9381