Ultimaker/Cura #12295: Does not save Directory location when you load a file, and does not name the file when you go to save it

CaryBab, 2022-05-21:
### Application Version

5.0

### Platform

Linux

### Printer

FLsun SR

### Reproduction steps

Loaded and save a file

### Actual results

Did not remember where I loaded file from when I went to load additional files.

Did not name the file by the name that it loaded, and did not use the custom printjob naming that was defined

### Expected results

All previous versions remembered where you loaded a file from.

All previous versions named the file to save automatically with the name you loaded, or used the custom printjob naming that was configured.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional Information

Vandresc, 2022-06-03:
Hi @CaryBab , thank you for your feedback! 
Devs see CURA-9261 , for our internal reference.

CaptainMalu, 2022-06-12:
I saw this also for files which will be directly printed via octoprint.
So you can't reprint from files stored on octoprint as files with same names (in this case no name) will overwrite each other.

peterborkuti, 2022-06-19:
Hi,
I think, some users asked for help regarding this part: Does not name the file when saving.
I attached some screenshots.

https://community.ultimaker.com/topic/40717-filenames-empty-in-file-portal

marthinwurer, 2022-07-01:
Trying to upgrade to 5.0 for the new slicing, ran into this. It's really annoying, and will probably keep me from using 5.0 for the time being. I'm on pop_os, but I'm pretty sure most ubuntu derivatives would have the same issue.

JesConsuegra, 2022-10-11:
I can confirm it also happens on MX Linux.

LarrySikes, 2022-10-22:
Same with me - this project have any quality control? Far too many bugs - going to supper slicer bye.

getglenn, 2022-12-02:
I have used Cura since way back and have had minor issues that were all resolved quickly, but the above issues have me looking at alternative slicers... I am on Cura 5.1.1 on Linux Mint 21 (see attached screenshot)
[image]

CaryBab, 2022-12-02:
I've just stayed at 4.13.1. Since they don't appear to be going to fix 5.x

AdrianLeal, 2022-12-03:
is there any update on that,, its very annoying have to select directory every time you open or save a file (also have to set filename on save) specially when you work with projects with many stl files which are on same directory and normally are saved on specific directory.

Vandresc, 2022-12-05:
Based on all your reports I re-discussed the priority with the team to see if we can pick it up sooner.
We do hear all of you :slightly_smiling_face:  Thank you for the reports, it helps us to make fixes.

Zotter, 2022-12-27:
FWIW - seeing this issue, still, with Cura 5.2.1 on Fedora 37.

As well as the lack of a file name when saving gcode. Have to find folders repeatedly, and have to fill in the file name - each time saved.

robertkardos, 2023-03-31:
I still have this issue, but the "remembering the file names" half seems to be kinda solved by [#12659](https://github.com/Ultimaker/Cura/issues/12659), though it works a little differently that it did in Cura 4, I explain it [here](https://github.com/Ultimaker/Cura/issues/12659#issuecomment-1491879512) a bit more in depth

TungstenX, 2023-06-24:
It seems to still be an issue on Ultimaker Cura 5.3.0. on Linux Mint 21.1

Gothicawakening, 2025-02-09:
Still an issue in 5.9.

It's somewhat ridiculous that a bug from 2 years ago that makes using Cura so frustrating has still not been fixed!