Ultimaker/Cura #12296: Bridging needs its own skin overlap setting & top surface skin should have its own skin overlap setting as well.

PhilBaz, 2022-05-21:
### Is your feature request related to a problem?

Yes, bridging skin overlap would be a functional non-aesthetic choice, normal skin overlap is  functional and aesthetic, top surface skin the choice would be aesthetic.

Currently these three choices are married by one setting. 

### Describe the solution you'd like

Separate skin overlap settings for 

1) skin
2) bridge
3) top surface skin

### Describe alternatives you've considered

There isn't any really....

### Affected users and/or printers

All

Erin-Stephan, 2023-04-24:
I am also interested in this. Not sure why there was no follow-up.

PhilBaz, 2023-04-25:
@Erin-Stephan 

I tend to use smartavionics branch 80% of the time. This feature was already implemented. 

https://github.com/smartavionics/Cura/releases/tag/4.20.11

Most feature requests go ignored with the main branch and the bugs have been just piling up since the 5.0 release. Meanwhile Prusa slicer is making serious progress. 

Starting to feel like maybe Cura's best days are behind it. mainbrach anyways. Check smartavionics branch. its a gold mine.

Erin-Stephan, 2023-04-25:
@PhilBaz  thank you. If a solution has been implemented by @smartavionics, couldn't it just be pulled into the main branch of Cura? Or is it more complicated than that? I would think if the work's already been done they could just use that same updated piece of code.

PhilBaz, 2023-04-25:
@Erin-Stephan I think there is a bunch of bureaucracy, maybe some egos involved.  He has implemented a ton of improvements and his version is rock solid. I have seen examples where they kind of ignore his solutions and features. He was previously a significant contributor to the main branch. Not sure if he still does.

Erin-Stephan, 2023-04-25:
Gotcha, ok