Ultimaker/Cura #12297: Changes to start gcode doesn't stay changed.

Windbane, 2022-05-21:
### Application Version

5.0

### Platform

Windows 11

### Printer

Ender 6

### Reproduction steps

Go to machine settings and make changes to start code.

### Actual results

Changes to start gcode doesn't stay changed. I'll go back to machine settings after closing window, and my start code doesn't have the changes I made previously. 

### Expected results

Start code edits should be saved.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8748242/cura.log)

Windbane, 2022-05-21:
I fixed the issue. The cura 5.0 folder in AppData\Roaming was set to "read only" for some reason. After unchecking "read only" and restarting cura 5.0, my edits to the start code remained changed.

closed 2022-05-21 by Windbane

Sonicboom247, 2022-05-21:
i noticed the same thing on mine, it was driving me crazy but i did not have to change folder permissions, for some reason i needed to make changes and stay on the settings page for a bit, then close and re open and they saved, but it is definitely not instant and super annoying that text highlighting is missing as well