Ultimaker/Cura #12306: Cura misbehaves with an uncommon dual screen arrangement

fast2611, 2022-05-22:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

I've got two different size screens using different element size settings in windows, in this arrangement.
[image]
Cura 4.13 did not have this issue.

### Actual results

Popups windows spawn in the "void" of no screens in the bottom right, thus being completly inacessible for all by the full screen size wide ones.
Cura also randomly switched from the correct UI element scale for the screen it was on to a significantly smaller one as has not reverted with system restarts

### Expected results

Cura needs to check if popups are inbounds and fully visible when creating them, or needs an option to be able to be locked to a single screen.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none needed

usernametakenaaaa, 2022-05-29:
At least it does not crash.
Try move the second screen down so the dialog can be rendered visibly, then it will crash.

Vandresc, 2022-05-31:
Thank you, @usernametakenaaaa, this is now on our radar. Duplicate of https://github.com/Ultimaker/Cura/issues/12102

closed 2022-05-31 by Vandresc

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

- A surprising connection was found to some particular audio drivers: https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232
- Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

MariMakes, 2023-03-31:
Quick update on our side,

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)