Ultimaker/Cura #12307: Application Menu on Mac is displayed inside the application window.

Spirou42, 2022-05-22:
### Application Version

5.0.0

### Platform

macOS Monterey 12.4

### Printer

Ultimaker 3

### Reproduction steps

Start application.

### Actual results

The application menu is displayed inside the application window.

### Expected results

Application menu is displayed in the mac menu bar

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

notzig the cause the is a platform related issue

Vandresc, 2022-05-31:
Hi, @Spirou42 , thank you for your feedback. Due to other high-priority issues we decided to defer this issue, meaning will stay Open but is not planned to be fixed yet.

thatcomputerguy0101, 2022-06-05:
Is there any specific reason why [QMenuBar](https://doc-snapshots.qt.io/qt6-6.3/qmenubar.html) was not used for the menu bar, or was it just not considered? Looking over the documentation, it seems to use the native menu bar in operating systems that have one, falling back on a custom implementation in other operating systems.

Vandresc, 2022-06-07:
@nallath could you please answer? Thank you!

fieldOfView, 2022-06-07:
QMenuBar was not used because the menu is implemented in QML using QtQuick controls. QtQuick 2 controls no longer have native versions (they did in QtQuick 1).

There's an alternative experimental native menu in QML which could be explored going forward:
https://doc.qt.io/qt-6/qml-qt-labs-platform-menu.html

nallath, 2022-06-08:
We had some issues getting the qt-labs one working, so at a certain moment we decided that it wasn't worth the effort to spend more time it at the time. We basically had the choice between having something at all and having a native menu bar.

Responses to the choice have been mixed. Some people like it, some people don't. It's up to our UX'er and PO what to do at this point.

nallath, 2022-06-08:
Sideloading a part of QT that isn't bundled is going to be an interesting challenge ;)

queengooborg, 2022-07-24:
(This appears to be a duplicate of #12244 -- I'm responding to this issue since the other one had a sour ending.  ;P)

I'd very much love to see this resolved soon.  As a native macOS user, it's sad to see a regression from native implementations, and I've personally stuck with Cura 4.13.1 (there are other UI changes that led me to stay on an older version).  As a developer, though, I understand that it's somewhat of an upstream problem -- it's disappointing that QtQuick has dropped support for native integrations like menu bars.

Anyways, just wanted to share my opinion on the topic. Wishing all the devs the best of luck on resolving this issue!

nanoant, 2022-07-28:
> We had some issues getting the qt-labs one working, so at a certain moment we decided that it wasn't worth the effort to spend more time it at the time. 

@nallath Can you please explain what was the problem exactly? Is this because PyQt6 does not ship with `Qt.labs.platform` module? (NOTE: PySide6 does, but I guess moving to PySide6 no go) Or is this something else?

I tried to look for `libqtlabsplatformplugin site:www.riverbankcomputing.com` but found absolutely nothing why they don't ship this module.

> Responses to the choice have been mixed. Some people like it, some people don't. It's up to our UX'er and PO what to do at this point.

I can hardly find positive responses, but easily can find negative ones:
https://www.reddit.com/r/Cura/comments/uwwr0l/cura_5_mac_not_behaving_properly_menu_bar/

I am (ex-)macOS developer and I'd gladly help to get the native menus back, but I'd need to understand what exactly is the problem to use `Qt.labs.platform`.

nallath, 2022-08-30:
I don't entirely remember what the problem was (it's been a few months and they have been pretty hectic).

From what I can remember is that the Qt-labs stuff didn't work on all operating systems. The wording that is used by Qt about labs also gives the impression that it's not entirely stable (although that could entirely be to prevent complaints/remarks about it not working)

probonopd, 2022-11-19:
This also affects other OSes which are using a global menu bar, such as [helloSystem](https://hellosystem.github.io/).

[image]

According to https://www.reddit.com/r/Cura/comments/uwwr0l/comment/i9urwky/?utm_source=reddit&utm_medium=web2x&context=3

> The reason this was changed is because in Cura 5.0, Cura updated to use Qt6 instead of Qt5 (primarily needed to support MacOS 11), and this dropped support of QtQuick Controls 
(...)
QtQuick Controls 2 has an application menu, but it doesn't support hotkeys any more then, such as Ctrl+O for open, Ctrl+S for save or Ctrl+P for print.

https://doc.qt.io/qt-6/qml-qtquick-controls2-action.html#shortcut-prop does make it look like you can actually have keyboard shortcuts on actions in QtQuick Controls 2?

VrtlBrown, 2023-11-01:
Circling back after a year to see if the status on this has changed at all?