Ultimaker/Cura #12312: Cura 5 on Mac error with slice

hendriksen-mark, 2022-05-22:
### Application Version

5.0.0

### Platform

MacOs Monterey 12.3.1

### Printer

cr-x pro

### Reproduction steps

1. add a stl
2. rotate
3. multiply selected
4. mirror 1
5. arrange all models
6. press slice

### Actual results

error:
Slicing failed with an unexpected error.

### Expected results

cura should finish slicing with no error.
cura 4.13.1 works fine

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8749666/cura.log)
[image]
[test.zip](https://github.com/Ultimaker/Cura/files/8749682/test.zip)

cura log and file that make the error

GT-Geo, 2022-05-22:
I'm using MacOs Monterey 12.3 and the latest Cura V5.0.0 and it sliced without errors for me.
[image]
.

nallath, 2022-05-23:
It does reproduce for me on Ubuntu 21 with 5.0.0.

Thanks for reporting this. I will discuss this with the other developers!

Ghostkeeper, 2022-06-16:
This occurs when Randomize Infill Start is enabled, but when there are layers without any infill lines (e.g. concentric, or just small infill parts).

closed 2022-07-11 by Vandresc