Ultimaker/Cura #12313: No support for multiple models

osterac, 2022-05-22:
### Application Version

5.0

### Platform

Windows 11

### Printer

Creality Ender 4

### Reproduction steps

I imported 2 models that both need support in cura. I selected one at a time mode and normal supports but this also happens with all at once mode. I sliced the models and took a look at the preview.

### Actual results

One of the models had support but one did not. If I use tree support, they both have support.

### Expected results

Both of the models should have support. 
[Drawer Wheel.zip](https://github.com/Ultimaker/Cura/files/8749723/Drawer.Wheel.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8749724/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This also happens in cura 4.13.1.

Vandresc, 2022-06-03:
Hi @osterac , thank you for your feedback. We discussed this case, and because of the size of the models in combination with the 0.2 Layer height, the support is indeed not generated. There are a few things you could adjust to generate support. There is Minimum Support Area that needs to be set to 0, which is actually the default value in our profiles, so this is a case that we wouldn't have seen. All together, Cura works as expected, therefore we will reject this ticket, I hope you understand.

closed 2022-06-03 by Vandresc

osterac, 2022-06-06:
>  Minimum Support Area that needs to be set to 0
Thanks, it works properly now. Not sure how that got set that way - it is an old profile I've been using for quite a while (since version 3).