Ultimaker/Cura #12316: Cura 5.0

sjbaird, 2022-05-22:
### Application Version

5.0

### Platform

Windows 10

### Printer

Tevo Trantula

### Reproduction steps

Have two extruders defined and make a change like nozzle diameter. Then click stettings menu option.

### Actual results

Duplicated extruder options, the first time was about 4 duplicates.
Once the mouse hovers above a selection the program crashes and all work is lost.

[image]

### Expected results

Not to crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

nallath, 2022-05-23:
Are you very sure that you're using 5.0.0 and not the 5.0 beta? This was an issue with the Beta, but has been fixed for the final 5.0 release. 

I've just re-tested this in the final 5.0.0 release and the issue doesn't happen there.

sjbaird, 2022-06-06:
Yep was beta, I hadn't realized 5 was released yet.

closed 2022-06-06 by sjbaird