Ultimaker/Cura #12318: Local Top/Bottom lines become sandwiched with infill

Obliman, 2022-05-22:
### Application Version

5.0.0

### Platform

Window 10

### Printer

Ender 5

### Reproduction steps

1. Slice a model **with or without** Alternate Extra Wall (AEW) enabled which produces intermediate Top/Bottom lines (for example, a shallow cut in a horizontal surface).
I've noticed strange behavior in both cases (AEW on and off), though it seems to be worse with AEW on.

### Actual results

Cura sandwiches infill (orange) between the Top/Bottom areas (yellow) instead of an additional Top/Bottom area.
Ex with AEW on:
Part 1: One extra infill layer
[image]
[image]

Part 2: Two extra infill layers
[image]

Ex with AEW off:
Part 2:
Note that there are more layers with the correct Top/Bottom fill than when AEW was off, but there is still a mystery infill layer sandwiched between those layers and the walls on the next layer.
[image]
[image]

### Expected results

The expectation is that only Top/Bottom is filled in the area, with no infill being sandwiched:

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It seems like this may only happen when the Top/Bottom fill area is a consistent shape between layers; when there is a slope I have not observed this issue, but it does appear for straight cuts.

[bug_example.zip](https://github.com/Ultimaker/Cura/files/8750256/bug_example.zip)

nallath, 2022-06-10:
Due to the length of our backlog and because this issue doesn't seem to adversely impact the quality so much, we've decided to defer this issue for now.

Obliman, 2022-06-10:
Thank you for the update. I have also not noticed any obvious impact on quality since these areas tend to be internal and covered by additional layers.

Obliman, 2023-06-11:
**Update per bot request:**
As of Cura 5.3.1 this is still an issue (I opened the project file in the original post and used its settings):
[image]

Obliman, 2023-06-11:
After some trial-and-error I found that this bug appears for this profile if `Wall Line Count * Line Width == 2`, so 2 lines for a 1mm line width, 4 lines for a 0.5mm line width, or 10 lines for a 0.2mm line, etc.
I assume there is some edge-case interaction with another setting, or maybe some kind of rounding error.

MariMakes, 2023-07-10:
Hey @Obliman,

Sorry, it took us a while to get back to you 😞 
Thanks for your udpate👍

I'm observing the same as you do, it also only seems to happen with a very specific wall line count. 🤔 
I'm also noticing one other thing, if you have Alternate Extra Wall, disabled it also working regardless of the wall line count. So that might be the culprit setting. 
[image]

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞