Ultimaker/Cura #12319: cannot save to Mac hard rive

go-bears, 2022-05-22:
### Application Version

5.0

### Platform

MacOS BigSur

### Printer

Ender 3 Pro

### Reproduction steps

open an stl file
select ender 3 pro printer
slice file
attempt to save the sliced file 
attempt to save the project
attempt to export the project

### Actual results

I tried to save to disk via the prompt
[image]

I tried to save project and tried to export the gcode file via the menu item
[image]

### Expected results

I should be able to have a file on my local hard drive. I can't save the project or the file, or export the file on the Mac laptop running BigSur. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

```
2022-05-22 14:26:19,922 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.04629206657409668 seconds
2022-05-22 14:26:53,811 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/bay-area-topo.gcode]...
2022-05-22 14:26:53,925 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [153]: Writing to Local File /bay-area-topo.gcode in text mode
2022-05-22 14:26:53,925 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /bay-area-topo.gcode: [Errno 30] Read-only file system: '/bay-area-topo.gcode'
2022-05-22 14:26:54,233 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:27:03,914 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.03630399703979492 seconds
2022-05-22 14:27:06,570 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/CE3PRO_rawmodel-94822.gcode]...
2022-05-22 14:27:06,684 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [153]: Writing to Local File /CE3PRO_rawmodel-94822.gcode in text mode
2022-05-22 14:27:06,684 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /CE3PRO_rawmodel-94822.gcode: [Errno 30] Read-only file system: '/CE3PRO_rawmodel-94822.gcode'
2022-05-22 14:27:07,033 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:27:26,828 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2022-05-22 14:27:26,829 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2022-05-22 14:27:26,859 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/shaders/xray.shader]...
2022-05-22 14:27:29,218 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PreviewStage
2022-05-22 14:27:29,227 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
2022-05-22 14:27:45,910 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.031891822814941406 seconds
2022-05-22 14:27:46,824 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/CE3PRO_rawmodel-94822.gcode]...
2022-05-22 14:27:46,933 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [153]: Writing to Local File /CE3PRO_rawmodel-94822.gcode in text mode
2022-05-22 14:27:46,933 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /CE3PRO_rawmodel-94822.gcode: [Errno 30] Read-only file system: '/CE3PRO_rawmodel-94822.gcode'
2022-05-22 14:27:47,232 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:30:19,080 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-05-22 14:30:32,067 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/bay-area-topo.gcode]...
2022-05-22 14:30:32,180 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [153]: Writing to Local File /bay-area-topo.gcode in text mode
2022-05-22 14:30:32,180 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /bay-area-topo.gcode: [Errno 30] Read-only file system: '/bay-area-topo.gcode'
2022-05-22 14:30:32,624 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:30:43,222 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-05-22 14:30:49,820 - WARNING - [MainThread] UM.Preferences.setValue [92]: Tried to set the value of non-existing setting asked_dialog_on_project_save.
2022-05-22 14:30:59,500 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/bay-area-topo.3mf]...
2022-05-22 14:30:59,615 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [156]: Writing to Local File /bay-area-topo.3mf in binary mode
[... 5 lines cut ...]
2022-05-22 14:31:27,963 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:53:9: QML ViewMenu: Binding loop detected for property "width"
2022-05-22 14:31:43,606 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-05-22 14:31:49,282 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to XRayView
2022-05-22 14:31:49,284 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/shaders/xray_composite.shader]...
2022-05-22 14:31:49,305 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/shaders/xray.shader]...
2022-05-22 14:31:49,315 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/shaders/xray.shader]...
2022-05-22 14:32:03,142 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
2022-05-22 14:32:43,920 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to XRayView
2022-05-22 14:35:39,435 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/bay-area-topo.gcode]...
2022-05-22 14:35:39,552 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [153]: Writing to Local File /bay-area-topo.gcode in text mode
2022-05-22 14:35:39,552 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /bay-area-topo.gcode: [Errno 30] Read-only file system: '/bay-area-topo.gcode'
2022-05-22 14:35:39,840 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:35:49,939 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.0333859920501709 seconds
2022-05-22 14:36:47,106 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-05-22 14:36:50,449 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [/bay-area-topo.3mf]...
2022-05-22 14:36:50,571 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [156]: Writing to Local File /bay-area-topo.3mf in binary mode
[... cut ...]
2022-05-22 14:37:05,055 - ERROR - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [180]: Operating system would not let us write to /bay-area-topo.gcode: [Errno 30] Read-only file system: '/bay-area-topo.gcode'
2022-05-22 14:37:05,273 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
2022-05-22 14:37:14,946 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.030163049697875977 seconds```

nallath, 2022-05-23:
I don't think that this is something that we can fix. As you can see from the logs, the operating system tells Cura that it's not allowed to write to the location that you have told it to write to.

Sciency-Owen, 2022-05-31:
I have had a similar problem. What I have found to work is that you cannot save on the Macintosh Disk as labelled that way, but can export the gcode to specific folders on the disk.

closed 2022-06-21 by github-actions[bot]