Ultimaker/Cura #12320: Fill empty cilinder

daredevil-ferreira, 2022-05-22:
### Application Version

5.5.0

### Platform

Windows 10

### Printer

CR10s PRo

### Reproduction steps

I upload the 3MF file.
After setting position to 0, 0, 0 and set the object upside down
I define to have supports on the buildtable only and have an infill of 100%

### Actual results

All the cylinders are filled up with material.

### Expected results

The Cylinders should remain empty as it is shown before slicing.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR10SPRO_Part Studio 1 - Part 1 (1).zip](https://github.com/Ultimaker/Cura/files/8750359/CCR10SPRO_Part.Studio.1.-.Part.1.1.zip)

nallath, 2022-05-23:
The model that you're using isn't manifold. When selecting all non-manifold edges in Blender, i get the following result:
[image]

This means that it's not a valid model to slice with in the first place. Cura can often fix these issues, but as you can see, this is not guaranteed.

closed 2022-05-23 by nallath

daredevil-ferreira, 2022-05-23:
I did create this file with onshape and i don't understand what is non manifold and how to avoid it.

For the time beeing, i will use superslicer for this model - it worked.

Regards
David