Ultimaker/Cura #12321: Cura 5 fails to slice a specific model

FacuM, 2022-05-22:
### Application Version

5.0.0

### Platform

Windows 11 21H2

### Printer

Creality Ender 2 Pro

### Reproduction steps

1. Import my profile
2. Import the attached model
3. Try to slice it

### Actual results

Cura crashes and asks me to fill a bug report.

### Expected results

Get the model sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE2_3110_Split_bottom_Dispenser_eggs.zip](https://github.com/Ultimaker/Cura/files/8750455/CE2_3110_Split_bottom_Dispenser_eggs.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8750458/cura.log)

GregValiant, 2022-05-23:
It didn't slice when I opened it.  Changing the Top/Bottom line width from .75 to 1.0 got it to slice.

FacuM, 2022-05-23:
> It didn't slice when I opened it. Changing the Top/Bottom line width from .75 to 1.0 got it to slice.

Yep, I figured that must be why too, since it's the only "weird" parameter I've tweaked recently.

The log reports that the engine couldn't resolve the expected paths for slicing.

`2022-05-22 20:21:00,152 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] [ERROR] transition on segment lies outside of segment!`

However, some other models (maybe smaller) resolve this setting just fine.

Vandresc, 2022-06-07:
Hi @FacuM , I could indeed reproduce as well. I will bring this up with the rest of the team. Thank you for your feedback!

FacuM, 2022-06-07:
> Hi @FacuM , I could indeed reproduce as well. I will bring this up with the rest of the team. Thank you for your feedback!

Thank you for your response, I'll be keeping an eye on my issue for updates!

rburema, 2022-11-15:
The project slices for me in 5.2.1 (installed) and from the current main branch (from source).

As such, I'm closing this -- Please tell us if this is not resolved for you, then we can re-open if necessary.

closed 2022-11-15 by rburema